Lora-net/LoRaMac-node #128: patch to fix SX1276GetTimeOnAir

yanfli, 2016-08-31:
According to formula from SX1276-DS-R4. SX1276GetTimeOnAir returns wrong value for SF11/SF12 low data rate optimization case. Below patch could fix it, and also fix several comment.

[image]

```patch
diff --git a/src/radio/radio.h b/src/radio/radio.h
index 383037d..584bc08 100644
--- a/src/radio/radio.h
+++ b/src/radio/radio.h
@@ -233,7 +233,7 @@ struct Radio_s
      * \param [IN] modem      Radio modem to be used [0: FSK, 1: LoRa]
      * \param [IN] pktLen     Packet payload length
      *
-     * \retval airTime        Computed airTime (us) for the given packet payload length
+     * \retval airTime        Computed airTime (ms) for the given packet payload length
      */
     uint32_t  ( *TimeOnAir )( RadioModems_t modem, uint8_t pktLen );
     /*!
diff --git a/src/radio/sx1276/sx1276.c b/src/radio/sx1276/sx1276.c
index 45b2998..996eab4 100644
--- a/src/radio/sx1276/sx1276.c
+++ b/src/radio/sx1276/sx1276.c
@@ -799,14 +799,14 @@ uint32_t SX1276GetTimeOnAir( RadioModems_t modem, uint8_t pktLen )
             double tmp = ceil( ( 8 * pktLen - 4 * SX1276.Settings.LoRa.Datarate +
                                  28 + 16 * SX1276.Settings.LoRa.CrcOn -
                                  ( SX1276.Settings.LoRa.FixLen ? 20 : 0 ) ) /
-                                 ( double )( 4 * SX1276.Settings.LoRa.Datarate -
-                                 ( ( SX1276.Settings.LoRa.LowDatarateOptimize > 0 ) ? 2 : 0 ) ) ) *
+                                 ( double )( 4 * ( SX1276.Settings.LoRa.Datarate -
+                                 ( ( SX1276.Settings.LoRa.LowDatarateOptimize > 0 ) ? 2 : 0 ) ) ) ) *
                                  ( SX1276.Settings.LoRa.Coderate + 4 );
             double nPayload = 8 + ( ( tmp > 0 ) ? tmp : 0 );
             double tPayload = nPayload * ts;
             // Time on air
             double tOnAir = tPreamble + tPayload;
-            // return us secs
+            // return ms secs
             airTime = floor( tOnAir * 1e3 + 0.999 );
         }
         break;
diff --git a/src/radio/sx1276/sx1276.h b/src/radio/sx1276/sx1276.h
index 06ccf62..6634d41 100644
--- a/src/radio/sx1276/sx1276.h
+++ b/src/radio/sx1276/sx1276.h
@@ -299,7 +299,7 @@ void SX1276SetTxConfig( RadioModems_t modem, int8_t power, uint32_t fdev,
  * \param [IN] modem      Radio modem to be used [0: FSK, 1: LoRa]
  * \param [IN] pktLen     Packet payload length
  *
- * \retval airTime        Computed airTime (us) for the given packet payload length
+ * \retval airTime        Computed airTime (ms) for the given packet payload length
  */
 uint32_t SX1276GetTimeOnAir( RadioModems_t modem, uint8_t pktLen );
 
```

Regimbal, 2016-09-29:
I have found the same error in sx1272.c file.
Will try to apply it tomorrow

mluis1, 2016-09-30:
Thanks for the feedback. Both comments are valid.
The issue only affected SF11 and SF12 datarates.

closed 2016-09-30 by mluis1