Lora-net/LoRaMac-node #129: Question on LoraJoin Req

SyedIbrahimAbbas, 2016-08-31:
According to the LoraWan Spec in chapter 7.1.2, it is mentioned that LoraJoin request message should use the predefined 6 channels and should follow 0.1% dutycycling even if the band allows 1% dutycycling for normal application. But the stack doesn't differentiate between Join request message and normal application for checking against dutycycle limitation.
        So, can I assume that, it is not mandatory. Can I use other frequencies of other subband for joining procedure, if gateway supports it?

djaeckle, 2016-09-02:
Hi SyedIbrahimAbbas,

thanks for the report. Which LoRaWAN Specification version are you refering to?

SyedIbrahimAbbas, 2016-09-06:
Hi I am currently using LoraWAN stack version 2015-08-07, V 3.3.

mluis1, 2016-09-23:
The full compliance with the LoRaWAN specification v1.0 is achieved with version 4.2.0 of the stack.
The latest version 4.3.0 is now compliant with version 1.0.1 of the LoRaWAN specification where the Join and normal channels have the same dutycycle restrictions, in plus only the 3 default channels are mandatory for the Join procedure.

We would advise you to move your stack version to the latest one.

closed 2016-09-23 by mluis1