Lora-net/LoRaMac-node #130: >=V410 Timerfunctions accuracy

DaniKoller, 2016-08-31:
When using a timer per example in a 200ms Intervall in the main application, the 5s OTA Joining Rx1 windows will start instead of 4.996s in 5.013s and therefore is no more working with DR5. There is a loss of 17ms in the timer handling ! 
I think the mixing of ms and ticks (488.28125us), specially with the RtcGetElapsedAlarmTime() which has an accuracy of +-1ms, and the substraction of this elapsed_ms (+-1ms) on TimerStart() will destroy the timing. Is that possible ?

GregCris, 2017-01-25:
Yes, the RTC is not meant to be used as a SysTick and it is used to optimize the events going to, or from, the low power mode. Reading the RTC internal registers is taking some time and with an IRQ every 200ms, you are are basically adding a very small delay every time and at the end you have a much longer delay. There is a trade off, you can either use the RTC and have a very low power but some inaccuracy in your system when used too often for short delays, or you can use a much faster time base but your system will consume more power.

mluis1, 2017-02-03:
Can we close this issue?

closed 2017-02-03 by DaniKoller