Lora-net/LoRaMac-node #131: LoRaWAN compliance failed on receiving RX2 Window

MichaelDevid, 2016-08-31:
Hi all,

I'm using the version 4.3.0 and I went for LoRaWAN compliance at IMST today but my hardware and code were not able to receive a downlink package during the RX2 window (second window). Receiving during RX1 worked fine. 

I use a 8 MHz oscillator for the STM32L1 instead of a 16 MHz and don't use a module but our own circuit. I did reconfigure the PLL to have a 32 MHz frequency. What could be a possible issue for not receiving anything during RX2? 

Thanks in advance! 

Michael

pdemil, 2016-09-01:
Hi Michael,

Did the PER RX2 compliance test fail on all the datarates?
It is not related to the v4.3 source code.

Check when receive window 2 is opened, you can use a logic analyzer and see a small voltage drop when the radio is in receive mode.

Best
Pieter

MichaelDevid, 2016-09-02:
Hi Peter,

Yes it did fail on all datarates. Every test where RX2 was used it failed a 100%. What do you mean it is not related to the V4.3 source code? Do you think it is more a hardware issue? 

I will do that test.

Thanks

Michael

pdemil, 2016-09-02:
It is most likely a timing issue (hardware or software). It is not related to V4.3 source code because we  have passed the compliance tests. Of course, it can be related to the source code on your board because you have different hardware.

MichaelDevid, 2016-09-04:
I made a new copy from version V4.3 and I only changed the pinout so it correctly interfaces with the SX1272 chip. The only difference between the hardware/software is that I use a STM32L152xE processor instead of a STM32L151xC/B. 

I noticed in this pdf from ST that there apparently is a difference between these two processors with the RTC. The processor were the stack has been written for is a ULPM and I use a ULPH which should be compatible but it's not. 

I looked at the timing of the RX2 window and it seems to be fine it opens 1 second after the RX1 and RX1 opens 1 second after TX. 

http://www.st.com/content/ccc/resource/technical/document/application_note/7a/9c/de/da/84/e7/47/8a/DM00025071.pdf/files/DM00025071.pdf/jcr:content/translations/en.DM00025071.pdf

mluis1, 2016-10-27:
We are currently adding the support for the NAMote72 end-device which is based on the STM32L152RC.

While we don't publish the full source code please find attached the rtc-board.c/h implementation that we have made and which looks to be functional on that MCU.

Please give it a try and let us know if it solves your issues. 
[rtc-board.zip](https://github.com/Lora-net/LoRaMac-node/files/556298/rtc-board.zip)

mluis1, 2016-11-04:
Can we close this issue?

MichaelDevid, 2016-11-04:
Yes to solve this issue we started using the ST implementation of your code: 
http://www.st.com/en/embedded-software/i-cube-lrwan.html

I have not tried you rtc-board implementation but I will wait for the official release.

closed 2016-11-04 by MichaelDevid