Lora-net/LoRaMac-node #132: SRV_MAC_LINK_ADR_REQ after a CFList when AdrCtrlOn is false

BBoulet, 2016-09-01:
Hello,

We have detected a issue when we execute a SRV_MAC_LINK_ADR_REQ after a CFList in a JOIN_Accept.

When we pass in this code : 

//CFList
if( ( size - 1 ) > 16 )
      {
            ChannelParams_t param;
           param.DrRange.Value = ( DR_5 << 4 ) | DR_0;

            for( uint8_t i = 3, j = 0; i < ( 5 + 3 ); i++, j += 3 )
           {
                  param.Frequency = ( ( uint32_t )LoRaMacRxPayload[13 + j] | ( ( uint32_t )LoRaMacRxPayload[14 + j] << 8 ) | ( ( uint32_t )LoRaMacRxPayload[15 + j] << 16 ) ) * 100;
                  LoRaMacSetChannel( i, param );
           }
     }

The new channels are set.
If we receive an SRV_MAC_LINK_ADR_REQ in the downlink just after the first uplink after the Join we pass in this code : 

if( ( AdrCtrlOn == false ) && 
  ( ( ChannelsDatarate != datarate ) || ( ChannelsTxPower != txPower ) ) )
        {   // ADR disabled don't handle ADR requests if server tries to change datarate or txpower
            // Answer the server with fail status
                  // Power ACK     = 0
                  // Data rate ACK = 0
                  // Channel mask  = 0
                  AddMacCommand( MOTE_MAC_LINK_ADR_ANS, 0, 0 );
                  macIndex += 3;  // Skip over the remaining bytes of the request
                  break;
            }

if this condition is true, the devices answer to the serveur with Channel mask as fail. in this specific case, the device should not answer to the serveur with the channel mask with the good value notifying that the channel list is ok ?

pdemil, 2016-09-02:
Why is the server sending an SRV_MAC_LINK_ADR_REQ when AdrCtrlOn is false? 
The mote flags must be like this: 
Flags : ADR : 0, ADRAckReq : 0

BBoulet, 2016-09-02:
Because we start to work with an operator which has his startup procedure even when AdrCtrlOn is false, there is no documentation about his functionnement, this is why we have detected this behavior.

Cordialement,
Boulet Benjamin
Hardware & Software Engineer
[image3344]
12 rue ampère, 38000 Grenoble
T. +33 (0)4 76 21 64 05
benjamin.boulet@wyres.fr<mailto:benjamin.boulet@wyres.fr> | www.wyres.eu

/P please don't print this e-mail unless you really need to.

djaeckle, 2016-10-17:
Hi BBoulet,

thanks for the report. If this condition is true, the server is trying to change the datarate and/or the tx power, which is not allowed if ADR is disabled. What you can do is to enable and disable channels in the channel mask, but only if you don’t change the datarate and the tx power which the node is using currently.
We agree that the LoRaWAN Specification is not clear at this point, but we think that the implementation is correct and that it provides as much features as allowed.

mluis1, 2016-10-21:
Can we close this issue?

BBoulet, 2016-10-21:
Hello, yes you can close.

closed 2016-10-21 by mluis1