Lora-net/LoRaMac-node #134: Clarifications

coljander, 2016-09-13:
Hi

We just need clarification on a couple of points which is causing a bit of debate in the lab.

1) In Eu mode can the DutyCycle calculations be reset for the 868.1/3/5 channel band on receipt of the Join Accept message or do we have to wait for the duty cycle to work itself out.  This isn't a problem if the gateway populates the optional channel frequency list, but if it doesn't and leaves only a single band available it can take a while before a packet can be transmitted.

2) Is FSK50 supported in the stack?  We failed on these tests in our certification but it wasn't a problem as the tests are optional.  When I looked at the code at the time it didn't seem possible for them to pass as the code didn't seem to enter a receive state.  This was on code version 4.1.  If it isn't supported are there plans on bringing it in?

Thanks

mluis1, 2016-09-13:
1) The DutyCycle must be applied for all transmissions. Thus, the calculations can't be reset once the JoinAccept has been received.

This is a requirement from the ETSI regulations.

2) The FSK50 is supported in the stack. For each release of the stack we go to the LoRa-Alliance certification houses and we verify that all mandatory and optional features pass the certification.

The readme.md file provides a change log indicating if the tests were successful or not. 
As you can see the version 4.1.0 had issues on getting the optional FSK50 downlinks to pass the certification. From version 4.2.0 the issue has been solved.

Please find below an excerpt of the changelog.

6. Changelog
-------------
2016-06-22, V4.3.0
* General
    1. This version has passed all LoRa-Alliance compliance tests.
    2. Update the MAC layer in order to be LoRaWAN version 1.0.1 compliant

2016-05-13, V4.2.0
* General
    1. This version has passed all LoRa-Alliance compliance tests.
    2. Update STM32L1xx_HAL_Driver version to 1.5. Update related drivers and implementations accordingly.

2016-03-10, V4.1.0
* General
    1. This version has passed all mandatory LoRa-Alliance compliance tests.

       *One of the optional tests is unsuccessful (FSK downlinks PER on Rx1 and Rx2 windows) and is currently under investigation.*
    2. Removed support for Raisonance Ride7 IDE (Reduces the amount of work to be done at each new release)

coljander, 2016-09-13:
Thank you for clarifying that.

mluis1, 2016-09-22:
Can we close the issue?

coljander, 2016-09-23:
Yes.  Please close it .

Thanks

Colin

closed 2016-09-23 by mluis1