Lora-net/LoRaMac-node #135: Is "RtcComputeTimerTimeToAlarmTick" really in ms?

Mehradzie, 2016-09-15:
Hi all,

I have a little bit of problem reading this section of the code in **rtc-board.c**.
To my knowledge, LoRaMac-node code uses RTC to do the timing on the background as all the Timer functions in **Timer.c** are forwarded to RTC methods.

There is a method in **rtc-board.c** called "`RtcComputeTimerTimeToAlarmTick`" which accepts a parameter of a unit milliseconds to calculate the next RTC alarm. (called `timeCounter`)

Inside this method, `timeCounter `values in milliseconds are compared and used with values in seconds such as part below, 
I have highlighted **three** points in the code using `<--- TIME IN comment.`

`static RtcCalendar_t RtcComputeTimerTimeToAlarmTick( TimerTime_t timeCounter, RtcCalendar_t now )   // <-- TIME IN ms unit
{
    RtcCalendar_t calendar = now;

    uint16_t seconds = now.CalendarTime.Seconds;
    uint16_t minutes = now.CalendarTime.Minutes;
    uint16_t hours = now.CalendarTime.Hours;
    uint16_t days = now.CalendarDate.Date;
    double timeoutValueTemp = 0.0;
    double timeoutValue = 0.0;
    double error = 0.0;
    
    timeCounter = MIN( timeCounter, ( TimerTime_t )( RTC_ALARM_MAX_NUMBER_OF_DAYS * SecondsInDay * RTC_ALARM_TICK_DURATION ) );

    if( timeCounter < 1 )
    {
        timeCounter = 1;
    }

    // timeoutValue is used for complete computation
    timeoutValue = round( timeCounter * RTC_ALARM_TICK_PER_MS ); // <-- TIME IN ticks units

    // timeoutValueTemp is used to compensate the cumulating errors in timing far in the future
    timeoutValueTemp =  ( double )timeCounter * RTC_ALARM_TICK_PER_MS;

    // Compute timeoutValue error
    error = timeoutValue - timeoutValueTemp;

    // Add new error value to the cumulated value in uS
    TimeoutValueError += ( error  * 1000 );

    // Correct cumulated error if greater than ( RTC_ALARM_TICK_DURATION * 1000 )
    if( TimeoutValueError >= ( int32_t )( RTC_ALARM_TICK_DURATION * 1000 ) )
    {
        TimeoutValueError = TimeoutValueError - ( uint32_t )( RTC_ALARM_TICK_DURATION * 1000 );
        timeoutValue = timeoutValue + 1;
    }

    // Convert milliseconds to RTC format and add to now
    while( timeoutValue >= SecondsInDay ) // <-- TIME IN ms units compared to time in seconds units.
    {
        timeoutValue -= SecondsInDay;
        days++;
    }`

Could somebody explain this to me please. It would be really appreciated,

Cheers,

Mehrad

adonislixi, 2016-09-18:
RTC is used to be a real time clock,  so one tik is one seconde after prescaler in normal way..
Here, lora use  Special frequency division (see board.c SystemClockConfig), it reach 2048 tik in one second.  so the clock is 2048 times faster than normal  clock..

Mehradzie, 2016-09-20:
@adonislixi 
I thought about it couple of times that maybe they are running the RTC faster but then i thought it is suppose to be REAL time clock. Nobody would do that :sweat_smile:

I can see that in the method that you pointed out they are setting the systick
`HAL_SYSTICK_Config( HAL_RCC_GetHCLKFreq( ) / 1000 );`

However, I don't have a full Keil license to run this code and from how I follow it up, `HAL_RCC_GetHCLKFreq( )` should read the value of the `uint32_t SystemCoreClock        = 32000000;` and return 32M accordingly. 

Which resolves into the final value below,

`HAL_SYSTICK_Config( 32000 );`

Would you please elaborate how you come up with the 2048 value?

Cheers
Mehrad

adonislixi, 2016-09-20:
Hi:
HAL_SYSTICK_Config( HAL_RCC_GetHCLKFreq( ) / 1000 ) is used to set the systick ,not rtc..
rtc use the LSE which is 32768Hz.
 See "rtc-borad.c, RtcInit", 
     RtcHandle.Init.AsynchPrediv = 3;
     RtcHandle.Init.SynchPrediv = 3;
32768 / (3+1)/(3+1) = 2048

adonislixi@foxmail.com
 
From: Mehradzie
Date: 2016-09-20 13:23
To: Lora-net/LoRaMac-node
CC: adonis lee; Mention
Subject: Re: [Lora-net/LoRaMac-node] Is "RtcComputeTimerTimeToAlarmTick" really in ms? (#135)
@adonislixi 
I thought about it couple of times that maybe they are running the RTC faster but then i thought it is suppose to be REAL time clock. Nobody would do that ????
I can see that in the method that you pointed out they are setting the systick
HAL_SYSTICK_Config( HAL_RCC_GetHCLKFreq( ) / 1000 );
However, I don't have a full Keil license to run this code and from how I follow it up, HAL_RCC_GetHCLKFreq( ) should read the value of the uint32_t SystemCoreClock = 32000000; and return 32M accordingly. 
Which resolves into the final value below,
HAL_SYSTICK_Config( 32000 );
Would you please elaborate how you come up with the 2048 value?
Cheers
Mehrad

mluis1, 2016-09-22:
In platforms where the SubSecond was nonexistent we had to accelerate the RTC peripheral in order to be able to count milliseconds instead of seconds. 

The drawback was that the SLEEP power consumption is a little bit higher (~3 uA if I remember well)

mluis1, 2016-09-30:
Can we close this issue?

adonislixi, 2016-09-30:
I thick so.

closed 2016-09-30 by mluis1