Lora-net/LoRaMac-node #136: Ack timeout

Kirankoneri, 2016-09-17:
Hi,

I am looking for a clarification on the necessity of ACK timeout event when the RX windows are enabled. As we know the Rx1 and Rx2 windows will be open based on Symbol timeout and receiver is turned off.  ACK timeout event occurs after (with 2 + rand (-1,1) seconds delay) the Rx2 window timeout. 

How does the ACK timeout event is helpful even after the RX2 window timeout.

Thank you,
Best Regards,
Kiran Koneri

mluis1, 2016-09-22:
The ACK timeout is useful in the situation where for some reason the radio missed a valid downlink on one of the reception windows.
This timeout ensures that the end-device does nothing for the maximum time on air of a valid frame.

Kirankoneri, 2016-09-26:
Why dont we consider the Rx2 timeout event as maximum time to ensure not to perform anything on end device. 
When we use Class A device, within Rx1 and Rx2 windows the end device can receive downlink. If the RX2 timeout happens then the mote does not have anything to do.

mluis1, 2016-10-27:
The ACK_TIMEOUT is used in order to start the same frame re-transmission as we didn't receive any Acknowledge on the 2 Rx windows.

Please find below the LoRaWAN specification paragraph that describes this.
> 
If an end-device does not receive a frame with the ACK bit set in one of the two receive windows immediately following the uplink transmission it may resend the same frame with the same payload and frame counter again at least ACK_TIMEOUT seconds after the second reception window. This resend must be done on another channel and must obey the duty cycle limitation as any other normal transmission. If this time the end-device receives the ACK downlink during its first receive window, as soon as the ACK frame is demodulated, the end-device is free to transmit a new frame on a new channel.

In case you don't agree with the specification I would advise you to address your concerns as well as your propositions to improve the situation to the LoRa Alliance Technical Committee. help@lora-alliance.org 

I close this issue as it is related to the specification and not to the firmware implementation itself.

closed 2016-10-27 by mluis1

nkaramolegos, 2018-12-03:
Does anybody know which value is usually used for the ACK_TIMEOUT?