Lora-net/LoRaMac-node #138: mote reset command cause mote node locks, no more transmissions

lucapizz, 2016-09-19:
Hi,

my trouble is repeatable under the following procedure.

I am using lib 4.3.0 and my node send confirmed messages every 30 seconds. I have set to 3 the retry.

I send the command "mote reset" to the NS (network server) so the sequence numbers (up and down link) starts from zero. Then I turn on my mote that starts to transmit.

I wait after the SF is adjusted from server to mote, but this may be unnecessary.

After 9 or 10 succesfull confirmed messages,  I send a "mote reset" command to NS (network server) so it restarts the downlink sequence counter.
Then my mote starts the retry because it does not recevie confirm ack.

When the uplink sequence number (mote continue to count on) reach about 24 or 25 the mote stops sending messages.

Moreover I have seen in the NS.log the "InvalidDelay" error some time after the retry. To me this seems like the mote send message too soon.

Repeating the procedure I have seen the MLMEconfirm.STATUS is in error while MCPS and MIB are OK. 

Please could someone help me to understand the problem?
I know that mote reset in unnecessary but to me this trouble hides some other subtle problem.

Thanks in advance.
Bye bye.

djaeckle, 2016-09-27:
Hi lucapizz,

thanks for your report. One question: if I understood the issue correctly, when you send the first "mote reset", you restart also the node, but not the second time?

lucapizz, 2016-09-27:
Hi Daniel,
yes, you have well understood.I first send a mote reset,then turn on the node (or restart it),then send another mote reset without any other action.
(please note that English is not my mother tongue, sorry)Lucapizz

djaeckle, 2016-09-27:
Hi lucapizz,

ok, thanks for the feedback. So the issue should have nothing to do with the mote reset. We will try to reproduce the issue on our desk and will come back to you.

lucapizz, 2016-09-30:
Hi Daniel,
any news? Have you reproduced the issue on your desk?
Please note that the issue happens after the restart from zero of the "downlink sequence counter" forced by "mote reset" command on servers. So there is something to do with it.
Bye bye.

djaeckle, 2016-10-06:
Hi lucapizz,

still in progress. 

> Repeating the procedure I have seen the MLMEconfirm.STATUS is in error while MCPS and MIB are OK. 

Which MAC command are you sending?

lucapizz, 2016-10-06:
Hi Daniel,
in the log file I see only one type of mac command, the content is "A1EHAGE" that change SF from 12 to 7. The gateway send it to the mote and get the following answer "Awc" in the log file.
This happens on the down sequence counter 7. After the sequence 9, I send the "mote reset" command and the mote does not receive anymore. So the mote change the SF back to 8 and the server sends again the mac command "A1EHAGE" but without answer "Awc". This go on up to the sequence 25 where the mote stop sending.

However I can send you the whole log file if you prefer. It is not so big (about 180Kb) because I have just one mote in the test system.
Let me know.

djaeckle, 2016-10-06:
Hi lucapizz,

I think a log would be helpful. You can upload it here or send it to lora@stackforce.de.

lucapizz, 2016-10-06:
ok, I have sent the nslog.txt file to lora@stackforce.de

djaeckle, 2016-10-06:
Thanks, I have received it.

djaeckle, 2016-10-10:
Hi lucapizz,

I was able to reproduce the issue. In my test case I performed the mote reset on the server manually.

After the reset, the server resets the downlink counter for it. When the node continues with uplink transmissions, the server responds with downlinks starting with the default counter value. 

The node will drop downlink frames until the maximum allowed counter difference is in a valid range (in this case, when the counters are equal). However, the node will drop this frame also, since it is duplicated. In your case, the server is sending confirmed downlinks. The node will drop the frame, but will not handle the internal state correctly, which prevents the node to transmit further uplinks.

The problem is related to issues #122 and #125. We have fixed it already on the develop branch. For me, a switch to the develop branch resolves the issue.

lucapizz, 2016-10-10:
Dear Daniel,
first of all thank you so much for your explanations.

Now I have loaded the "develop branch" version and my node continue to send uplink messages (it does not stop anymore).
So the issue could be closed.

But I would have another question. Tell me please if you may answer here or else where.

If I have well understood, when the server (for some reason) restarts the downlink conuter, we lose the communication with all nodes until such counters get back to a valid value. This may take very long time.

So, in such a situation, the node how must behave? Which policy the node should use?
Thanks in advance.

djaeckle, 2016-10-11:
Hi lucapizz,

The receiver of a LoRaWAN frame will drop a frame, if the counter difference of the stored counter with respect to the counter in the received frame is greater as `MAX_FCNT_GAP` (value is region specific - e.g. for EU, its 16384).

I think if you perform a mote reset on the server you will have to find a way to inform the node of the reset of the server. Otherwise, as you already said, the node discards frames until the counters are synchronized. However, maybe the question is why you need a mote reset on the server without a trigger from the node.

Most probably, LoRaWAN Specification 1.1 will come with a new MAC command called `ResetInd/ResetConf`. The node can use this MAC command to notify the server that it will perform a reset.

lucapizz, 2016-10-11:
Hi Daniel,

thank you very much for your help and your patience and your advices.

closed 2016-10-11 by lucapizz