Lora-net/LoRaMac-node #140: about class_b question

yanzi418, 2016-09-27:
In the class_b node code (feature), how to make nodes into class_b? Start beacon_pkt_fwd gateway. Now test node or class_a.

djaeckle, 2016-09-27:
Hi yanzi418,

thanks for your question.

We have a sample application which provides class b functionality for the LoRaMote platform. All other class b examples have not been updated yet. Please note, that we have only tested it for EU868 band. The class b sample application performs the following operations:

- It joins the network via APB or OTAA
- It performs an uplink containing the MAC command BeaconTimingReq.
- It performs a beacon acquisition. If the server answered the previous command, the node takes the response information into account. If not, it just transits in RX mode and "searches" a beacon for the time of a beacon period.
- If the node received a beacon successfully, it performs another uplink containing the MAC command PingSlotInfoReq. The node must receive a response from the server, otherwise it is not allowed to switch to class b. If the node received a response, it switches its operation mode to class b. If not, it starts the beacon acquisition procedure again. In the uplink, the additional MAC command LinkCheckReq does not effect the beacon operation.
- Once in beacon mode, the node receives beacons automatically. Please note, that the LoRaWAN Specification v1.1 defines some guard times around the beacon reception time. During this time, the node cannot perform any other uplinks. Also, the node opens ping slot windows (the periodicity depends on the configuration of the previously performed MAC command PingSlotInfoReq) for class b downlinks.

During operation, the node will notify the application if it has locked the beacon or not. If the node has not received a beacon for 120min, it will notify the application, such that it can switch back to class a.

In general, please note that the status of the implementation of class b is still experimental.

yanzi418, 2016-09-27:
Thank you, I would like to ask a question, class_b mode is related to the server? Do you need a server to process the beacon discovery command BeaconTimingReq? If the server is not the latest version of the server, you can test class_b? I now use the gateway is beacon_pkt_fwd kerlink.

adonislixi, 2016-09-29:
hi.. I found that in master , the application code in classA/**/main.c is the same as ClassB/**/main.c. As I understand it, there no stable classB code??  
I want to achive that switching A to B , A to C .. do I need use the classB in feature code ??

djaeckle, 2016-10-06:
Hi adonislixi,

currently, the class b implementation is on the feature/class_b branch. Please note, that we implemented it against the LoRaWAN Specification 1.1, which is still experimental and not released yet. Thus, the implementation is not finalized yet. If you want to use the class b feature, you will have to checkout the feature/class_b branch. The branch also supports class a and class c.
The master and the develop branch support class a and class c.

closed 2016-10-14 by mluis1