Lora-net/LoRaMac-node #143: RSSI and SNR

coljander, 2016-09-29:
Occasionally we are seeing weird RSSI and SNR values being reported.  Although probably not related to our strange values, when looking at the code and I noticed a couple of things that I'm hoping you can clarify.
1) In SX1272OnDio0Irq() the snr value that is eventually passed to OnRadioRxDone() is the value that is stored in the register and not the actual snr value.  You use SX1272.Settings.LoRaPacketHandler.SnrValue, but the actual value appears to be stored in the snr variable, which is only ever used in the RSSI calculation.  Am I reading that correctly?  Up to now we've been reporting the SX1272.Settings.LoRaPacketHandler.SnrValue as the actual snr value which seems incorrect.  
2) In the same function, in the calculation for the Rssi value when snr < 0, you've used the 16/15 scaling factor which is suggested in the SX1272 spec.   However the spec says that this scaling should only be applied when snr >=0.  Does this make a big difference to the reported RSSI value for weak signals?

Thanks

mluis1, 2016-09-30:
1) Please take a look at Issues #75 and #104. 
2) For weak signals, it does not make a lot of difference, since RegPktSnrValue is small (hence the 16/15) has low effect. However, what the specification may overlook is the situation where RSSI is high and SNR is negative (high interference and signal below interference). In this case it is better to still apply the 16/15 factor, in order to have a better evaluation of the level of interference.

coljander, 2016-10-03:
Hi

Thanks for clearing this up.  The issue can be closed.

Colin

closed 2016-10-03 by mluis1