Lora-net/LoRaMac-node #144: Sx1272 won't go into transmit mode when using LoRa

Thaskippy, 2016-09-30:
Hey,
I'm adapting the LoRaMac-Node v4.2 code for use on an ATxMega256A3U.
I'm trying to send a basic message in LoRa through the tx-cw example, however the sx1272 won't go from Standby into transmit mode.
When I use the FSK example from ["Closed issue 65"](https://github.com/Lora-net/LoRaMac-node/issues/65) everything works as expected.

Are there any special considerations before the sx1272 can switch from standby to transmit, that are not accounted for in the tx-cw example?

joeknapp, 2016-09-30:
The change between FSK and LoRa mode can only be done while in the Sleep state.

Thaskippy, 2016-10-01:
Thank you for the reply.
I do understand that I can only switch from FSK to LoRa when the sx1272 is in SLEEP mode. What I mean is that the sx1272 is already in STANDBY mode and in LoRa mode. In other words the value in register 0x01 is equal too 0x81. 

When I use the tx-cw example provided by the LoRaMAC-Node source then it will enter LoRa STANDBY mode, do all the register changes as expected, but then it will not go into LoRa TRANSMIT mode even after requesting through writing 0x83 to register 0x81.

mluis1, 2016-10-03:
The tx-cw example is meant to transmit a continuous wave only. This mode of operation is useful for ETSI, FCC, or other national organism radio certification. The continuous wave goal is just to have a non modulated tone at the center frequency.

The radio will never transmit useful data when the continuous mode is activated.

If you want to transmit data you should use the Ping-Pong or the LoRaMac examples as a base for your trials.

Thaskippy, 2016-10-03:
Alright. Well I started using the tx-cw because I couldn't get the LoRaMac one to work either. I'm using the ping-pong example but it just won't go into transmit mode when using the LoRa configuration. I'm running the code below:

```
uint8_t Buffer[] = {'P', 'I', 'N', 'G'};
uint8_t BufferPktLen = sizeof Buffer;
	
SX1272Init(NULL);
                                                         
Radio.SetChannel( RF_FREQUENCY ); //  868 MHz, define taken from the ping-pong source	

Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                   LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                   LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                   true, 0, 0, LORA_IQ_INVERSION_ON, 3000 ); // Defines taken from the ping-pong source	
	
Radio.Send( Buffer, BufferPktLen );
	
while(1)
{
    while(1)
   {
       SX1272Read(REG_LR_OPMODE);  // Check if the operational mode changes into transmit
       SX1272Read(REG_LR_IRQFLAGS); // Check if TxDone IRQ is issued
   }
}
```

Am I right in assuming that with the above program the sx1272 should go into transmit and send the 4 characters PING in Buffer? Cause I'm not seeing it happening. The RXTX line and DIO0 line stay low and the register 0x01 stays put in 0x81 even though a request has been send to the sx1272 to go into transmit mode.

I've attached a logic analyzer to the SPI bus and the RXTX and DIO0 of the sx1272. The SPI communication is working as expected and when I use the aforementioned FSK setup from ["Closed issue 65"](https://github.com/Lora-net/LoRaMac-node/issues/65) I do see the RXTX line pulled high and a while later the DIO0 line, so my testing set-up should work as expected too.

mluis1, 2016-10-03:
Could you please try the PingPong example as is. If no other device is answering the PingPong example just restarts the "PING" message transmission after the Rx timeout.

In your example if you remove the call to `SX1272Read(REG_LR_OPMODE);` I think that your code will work better. It is better to not poll the OpMode register.

In order to make your example work I would modify your while loop as follows (please note that I didn't test the below example)
```C
uint8_t Buffer[] = {'P', 'I', 'N', 'G'};
uint8_t BufferPktLen = sizeof Buffer;
uint8_t irqFlags = 0;

Radio.Init( NULL );

Radio.SetChannel( RF_FREQUENCY ); //  868 MHz, define taken from the ping-pong source   

Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                             LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                             LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                             true, 0, 0, LORA_IQ_INVERSION_ON, 3000 ); // Defines taken from the ping-pong source 

while( 1 )
{
    irqFlags = SX1272Read( REG_LR_IRQFLAGS );
    if( ( irqFlags & RFLR_IRQFLAGS_TXDONE_MASK ) == RFLR_IRQFLAGS_TXDONE )
    {
        // Tx Done
        Radio.Sleep( );
        // Toggle a pin to indicate that the transmission has finished.
    }
}
```

I also think that you should initialize the radio using the interrupts callbacks.
```C
    // Radio initialization
    RadioEvents.TxDone = OnTxDone;
    RadioEvents.RxDone = OnRxDone;
    RadioEvents.TxTimeout = OnTxTimeout;
    RadioEvents.RxTimeout = OnRxTimeout;
    RadioEvents.RxError = OnRxError;

    Radio.Init( &RadioEvents );
```

Thaskippy, 2016-10-10:
Update in case anyone is wondering:
I've tried my code on a different stand-alone radio module; RF-LORA-868-SO LORA TRANSCEIVER by RF-Solutions.
With this module everything works as expected.

This means that my problem lies not in the LoRaMAC-node software, but is related to the hardware originally used (or perhaps lower level software). If anyone has an hardware related suggestion what could cause the SX1272 not to go into transmit-mode that would be greatly appreciated. 

Otherwise I geuss the topic can be closed.

Thanks for the help.

closed 2016-10-14 by mluis1

frank651, 2016-11-21:
Hi
I have same problem, my lora module is PM402 from manthink company. 
Would you please help me?

Thaskippy, 2016-11-22:
Hey Frank,
For me the problem with the sx1272 turned out to be a problem with the power supply.
Without a decent power supply it was apparently getting fed through the SPI lines which made it appear as though the chip was working correctly up until the point it actually needed a decent power supply and switch towards transmitting information.

Now since you're using different hardware I doubt you have the same issue. It is something you could look into however.

frank651, 2016-11-22:
Hi Thaskippy
Thanks so much for your replay.
I've tested with different power supply but still having problems, when switch on FSK mode everything works such as RXTX pin has changed, go to transmit mode, IRQ flags, .., but in LORA mode doesn't work anything.
I think my LORA module has problem, i contact with manthink company about this problem but them did not give me technical support, i really suggest to all do not use from manthink company products.

mingfeizhao, 2016-11-22:
Hi Frank.
             this is mingfei Zhao from ManThink technology. I am so sorry to hear about  your comment about our product. I think that maybe there is some misunderstand between you and ManThink. belows are some information about our product.
            1. PM402 is designed for Chinese market which can only support 410MHz-510MHz which is different with SX1272.
            2. Users can't design a product with LoRaWAN compliant , because PM402 is designed for a scenario which need a very high receive sensitivity which can achieve to -148dBm with 18bps. 
            3. also we have products with LoRaWAN compliant which have already got certification of LoRaWAN by 7layer.
            4. we never receive any information about your issue, there are some reseller on internet who sells  our product but  they can't supply any technique support. please confirm that if the guys you have contacted with  is from ManThink.
           5. if you are still interesting in our PM402. please give me a email to zhaomingfei@manthink.cn , I will give you reference codes to help you run up the module. before that, please confirm the PM402 is adapted to your project.
           6. if you are interesting in our LoRaWAN system, please contact with me, we can supply the total solution which include module , gateway and server.
           
           hope the information is useful to you. if any other question, please contact with me by zhaomingfei@manthink.cn.

frank651, 2016-11-22:
Hi Zhao

really thanks for your reply, you've changed my mind about your company, **NOW** i think manthink is professional company. 

**1. PM402 is designed for Chinese market which can only support 410MHz-510MHz which is different with SX1272.**
Yes, i Know about this, Thaskippy using SX1272 but i Know PM402 using SX1278 chip.

**2. Users can't design a product with LoRaWAN compliant , because PM402 is designed for a scenario which need a very high receive sensitivity which can achieve to -148dBm with 18bps.**
i don't need LoRaWAN, we must design especial protocol for our project.

**3. also we have products with LoRaWAN compliant which have already got certification of LoRaWAN by 7layer.**
Yes, i saw your certification in lora alliance, That's why I chose your product.

**4. we never receive any information about your issue, there are some reseller on internet who sells our product but they can't supply any technique support. please confirm that if the guys you have contacted with is from ManThink.**
I contacted with Liese from manthink

**5. if you are still interesting in our PM402. please give me a email to zhaomingfei@manthink.cn , I will give you reference codes to help you run up the module. before that, please confirm the PM402 is adapted to your project.**
Really thanks Zhao, Be sure to call you for my project.

## Hey guys manthink a nice company, Zhao changed my mind about manthink, i suggest to all See the [website](www.manthink.cn/en/ ) of the company

emq1, 2016-12-09:
,
Even I am facing same issue with SX1276 when trying to put modem in transmit mode(LORA).
It works fine with FSK, but in LORA I can only switch between stand by and sleep mode.
Other register configurations works fine with LORA mode.

Please let me know if anyone has got resolution for this issue.

Thaskippy, 2016-12-11:
Well have you checked the power supply? It should be stable around 3V.

emq1, 2016-12-13:
Hi Thaskippy,
Thanks for the reply. Yes I have checked power supply not an issue, It works fine in FSK mode, but when put in to LORA mode nothing works.

emq1, 2017-01-09:
If anybody is facing same issue see if this helps.
In my case PLL locking was the issue, for that when we debug found that there is mismatch in the crystal.
It is recommended to use 32 Mhz crystal, by mistake we were using 16Mhz.
Once this is corrected, LORA TX and RX modes are working fine. Able to transmit and RX bytes now.
Now starts the actual project..:)
cheers emq1

ios198, 2018-01-03:
I am trying to test Tx-Rx with SX1276.
This is my source code in TX for sending 'PING" continuously :
`

Radio.Init( &RadioEvents );
Radio.SetChannel( RF_FREQUENCY ); 
uint8_t Buffer[] = {'P', 'I', 'N', 'G'};
uint8_t BufferPktLen = sizeof Buffer;
uint8_t irqFlags = 0;
Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                             LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                             LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                             true, 0, 0, LORA_IQ_INVERSION_ON, 3000 ); // Defines taken from the ping-pong source 
Radio.Send( Buffer, BufferPktLen );
while( 1 )
{
    irqFlags = SX1276Read( REG_LR_IRQFLAGS );
    if( ( irqFlags & RFLR_IRQFLAGS_TXDONE_MASK ) == RFLR_IRQFLAGS_TXDONE )
    {
        // Tx Done
        Radio.Sleep( );
        DelayMs( 1 );
        Radio.Send( Buffer, BufferPktLen );
        // Toggle a pin to indicate that the transmission has finished.
    }
}`
This is my code in RX to receive data:
`

Radio.Init( &RadioEvents );
Radio.SetChannel( RF_FREQUENCY );
uint8_t irqFlags = 0;

Radio.SetRxConfig( MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
                               LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
                               LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
                               0, true, 0, 0, LORA_IQ_INVERSION_ON, true );
Radio.Rx( RX_TIMEOUT_VALUE );
while( 1 )
{
    irqFlags = SX1276Read( REG_LR_IRQFLAGS );
    
    if( ( irqFlags & RFLR_IRQFLAGS_RXDONE_MASK ) == RFLR_IRQFLAGS_RXDONE )
    {
        // Rx Done
        Radio.Sleep( );
        // Toggle a pin to indicate that the transmission has finished.
         DelayMs( 1 );
         Radio.Rx( RX_TIMEOUT_VALUE );
    
    }
}`
==>RxDone Interrupt occurred in RX when each 'PING' is sent from TX. However, received data is always 0, dataSize = 0.
Could you help me to solve this problem?

Josbleuet, 2018-05-04:
I have the same problem. I am using modtronix module with sx1276 chip. I tested 15 modules, 9 are working fine and 6 having the issue. Power supply is two AA batteries.
For the bad modules, 
In lora mode, 17 dbm Tx is not firing.
In lora mode, 5dbm tx is firing but received playload size is 0 followed by rxTimeout interrupt.

Josbleuet, 2018-05-04:
I also powered a module with a 3.3v regulator. No change, problem is still present. Voltage is stable even during a tx. (tested with a scope). No voltage drop.

killdaclick, 2019-10-07:
Same issue here on Murata (SX1276 + STM32):
For some time after powerup TX, RX and SLEEP working fine. After some packets send radio stuck in TX mode even nothi g is transmitted. Higher current is draw from power supply. It is possible to switch to RX mode and then again to TX but itis not possible to switch to SLEEP anymore. Write and the read the register confirms that module will not switch. Restart of the module helps for some time.