Lora-net/LoRaMac-node #146: UartMcuPutChar 不能发送数据。

Hellogz, 2016-10-14:
根据ST库描述：At transmission end of transfer HAL_UART_TxCpltCallback is executed and user can add his own code by customization of function pointer HAL_UART_TxCpltCallback，只有在发送数据后才进入HAL_UART_TxCpltCallback方法，而代码是先开启中断后，在HAL_UART_TxCpltCallback里进行发送数据，所以这样是无法发送数据出去的，这里的代码应该要与旧版本库的使用方法区别开来。
还有HAL_UART_TxCpltCallback方法里的HAL_NVIC_DisableIRQ( USART1_IRQn )，执行后将无法继续接受数据，因为没有地方再次开启USART1中断了。
修改后的代码：
`uint8_t UartMcuPutChar( Uart_t *obj, uint8_t data )
{
    if( IsFifoFull( &obj->FifoTx ) == false )
    {
        __disable_irq( );
        FifoPush( &obj->FifoTx, data );
        __enable_irq( );
        // Enable the USART Transmit interrupt
        __HAL_UART_ENABLE_IT( &UartHandle, USART_IT_TXE );
        HAL_UART_Transmit_IT( &UartHandle, &data, 1 );
        return 0; // OK
    }
    return 1; // Busy
}

void HAL_UART_TxCpltCallback( UART_HandleTypeDef *UartHandle )
{
    uint8_t data;

    if( IsFifoEmpty( &Uart1.FifoTx ) == false )
    {
        data = FifoPop( &Uart1.FifoTx );
        //  Write one byte to the transmit data register
        HAL_UART_Transmit_IT( UartHandle, &data, 1 );
    }
    else
    {
        // Disable the USART Transmit interrupt
        __HAL_UART_DISABLE_IT( UartHandle, USART_IT_TXE );
    }
    if( Uart1.IrqNotify != NULL )
    {
        Uart1.IrqNotify( UART_NOTIFY_TX );
    }
}
`
如果有错误，请指出。thank you .
by Hellogz

mluis1, 2016-10-14:
Could you please describe your issues using English. We don't understand Chinese (I think it is Chinese).

Hellogz, 2016-10-14:
my english is 0, please use google translation.

pdemil, 2016-10-14:
@Hellogz maybe you can use google translate and copy the translation here...

mluis1, 2016-10-14:
Several issues have already been opened concerning the Uart driver implementation. (#133, #110, #96, #79, #40).
Maybe the answer you are looking for is already answered in there.

For your information we are currently trying to improve the Uart driver implementation. An update for it should be available soon on the develop branch.

closed 2016-11-11 by mluis1

twowinter, 2016-12-09:
Hi cool friends,

I got the same problem when i need the UART1 for printf something.
I do a little test at PingPong example with SensorNode Project, my test code in 'board.c' is commited on github.[https://github.com/twowinter/LoRaMac-node/commit/2e7d18f1bbb95f84680aaec0166c5e0339c48738](url)

Could someone check my code, now the code is always stay in **USART1_IRQHandler**.

Hellogz, 2016-12-22:
@twowinter Hi, your github code notfound .

twowinter, 2016-12-22:
@Hellogz 
Hi, please copy the link, the github can't jump directly.
https://github.com/twowinter/LoRaMac-node/commit/2e7d18f1bbb95f84680aaec0166c5e0339c48738

it is my test code commit.

Hellogz, 2016-12-23:
@twowinter your code can't working? I encountered the problem of using the serial output garbled, because my MCU crystal oscillator for 16MHz reasons.

twowinter, 2016-12-23:
Is your MCU is STM32L151C8 just the same as SensorNode Platporm?
---------
if your  crystal oscillator is 16MHz, you can change my test project follow these steps(if these steps for change HSE is right). Just to check my test commit is right on your MCU:

1. \src\boards\SensorNode\cmsis\stm32l1xx_hal_conf.h

`#if !defined  (HSE_VALUE) 
  //#define HSE_VALUE    ((uint32_t)12000000) /*!< Value of the External oscillator in Hz */
  #define HSE_VALUE    ((uint32_t)16000000) /*!< Value of the External oscillator in Hz */
#endif`

2. \src\boards\SensorNode\board.c
`RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL6; //RCC_PLL_MUL12`

3. MDK Project setting
Option for Target: Xtal: 12MHz   ->   16MHz