Lora-net/LoRaMac-node #147: OTAA Join Request Problem

Hellogz, 2016-10-14:
如果第一次入网请求没有成功，节点设备会一直重复发送相同的入网请求数据，这样的行为不符合LoRaWAN协议：
> DevNonce is a random value.2 For each end-device, the network server keeps track of a certain number of DevNonce values used by the end-device in the past, and **ignores join requests with any of these DevNonce values from that end-device**.

yanzi418, 2016-10-14:
修改一下Mac状态查询就可以了，不让发同一包Join命令

mluis1, 2016-10-14:
Could you please describe your issues using English. We don't understand Chinese (I think it is Chinese).

Hellogz, 2016-10-14:
my english is 0, please use google translation.

mluis1, 2016-10-14:
We could use google translate. But, it would be nice if you post your comments directly in English by using yourself Google translate. Like this everyone reading your posts doesn't need to perform that task.

mluis1, 2016-10-14:
If we have understood it correctly we think that this issue is a duplicate of the issue #114 which has already a fix in place on the develop branch. Commit 7143d85f01643277880d17cb56f274bccb868537

Hellogz, 2016-10-14:
I'm using branch master,  but the problem still exists。

mluis1, 2016-10-14:
What do you mean by the problem still exists?

The Master branch only contains released versions which have gone trough the LoRa-Alliance protocol certification test houses procedure.

The develop branch contains the work preparing the next version of the software and we believe that the issue you describe is already corrected on the develop branch.

If you think that your issue isn't corrected on the develop branch then please try to provide a better description of the issue as well as more details on how to reproduce it.

Hellogz, 2016-10-14:
After I tried, yes the develop branch is fix the bug. ScheduleTx() change to OnTxDelayedTimerEvent().

mluis1, 2016-10-14:
Ok, thanks for the feedback. I close the issue.

closed 2016-10-14 by mluis1

Hellogz, 2016-10-20:
Mac状态查询?哪个位置？
开发版分支不是这样解决的。把改ScheduleTx( );为OnTxDelayedTimerEvent( );即可解决。