Lora-net/LoRaMac-node #148: Build project for STM32L0xx device

iBrick, 2016-10-16:
I see HAL drivers in project for STM32L0xx. But there's no config how to change original driver for STM32L153 to STM32L0 (L062 in particular). Could you give me a hint how to change driver in project?

iBrick, 2016-10-20:
Anyone?

littleboot, 2016-10-20:
In order to port the driver to another MCU you will have to understand how to interface between with the lower lever hardware and LoRaWAN is implemented. Hint: read and understand the driver. Also I believe ST provides a application note about porting firmware from Cortex M3 to M0 and the things you need to consider. (I will be porting the firmware to an Cortex M0 in a few months, but I probably won't be able to share this firmware, I will have to discuss this with my customer first. Personally I would like to share the firmware because open source will increase the quality)

iBrick, 2016-10-20:
Thank you, I got it. I was hoping it can be done via confuguration, but no luck =)

mluis1, 2016-10-20:
We are currently adding a new platform based on the STM32L051C8. This work will be pushed to the develop branch in the coming weeks once we have something more stable.

littleboot, 2016-10-20:
Hello thank you for letting me know. I'm very happy with your response because the STM32L051C8 is the MCU I chose to use inside my project too! I hope I can contribute to the code when it's available.

mluis1, 2016-11-29:
We close this issue as we just pushed the MoteII platform support which is based on the STM32L051C8 MCU.

closed 2016-11-29 by mluis1