Lora-net/LoRaMac-node #149: Issues with Sticky mac answers

Mobiak, 2016-10-20:
There is two issues possible with the sticky MAC answers:
1 - The code implemented as an error in the number of payload for the answer MOTE_MAC_RX_PARAM_SETUP_ANS (2 instead of 1). It will take one more Byte on the next command or take a number present in the buffer.
2 - If there is another command it will be discarded, but not the payload associated. If this payload is equal to one of the command generating a sticky process, it will generates some false answers to repeat.

This is the case in the develop branch commit f435b5412a084b2beac02851459367f0fec1336f and was initiated in the commit  d2c1bbf5d9b7d6b19e1cd90bf6aa01f402bd08f3 between release 4.2.0 and 4.3.0.

The actual code is :

```
static uint8_t ParseMacCommandsToRepeat( uint8_t* cmdBufIn, uint8_t length, uint8_t* cmdBufOut )
{
    uint8_t i = 0;
    uint8_t cmdCount = 0;

    if( ( cmdBufIn == NULL ) || ( cmdBufOut == NULL ) )
    {
        return 0;
    }

    for( i = 0; i < length; i++ )
    {
        switch( cmdBufIn[i] )
        {
            case MOTE_MAC_RX_PARAM_SETUP_ANS:
            {
                cmdBufOut[cmdCount++] = cmdBufIn[i++];
                cmdBufOut[cmdCount++] = cmdBufIn[i++];
                cmdBufOut[cmdCount++] = cmdBufIn[i];
                break;
            }
            case MOTE_MAC_RX_TIMING_SETUP_ANS:
            {
                cmdBufOut[cmdCount++] = cmdBufIn[i];
                break;
            }
            default:
                break;
        }
    }

    return cmdCount;
}
```

A correction could be something like the following

```
static uint8_t ParseMacCommandsToRepeat( uint8_t* cmdBufIn, uint8_t length, uint8_t* cmdBufOut )
{
    uint8_t i = 0;
    uint8_t cmdCount = 0;

    if( ( cmdBufIn == NULL ) || ( cmdBufOut == NULL ) )
    {
        return 0;
    }

    for( i = 0; i < length; i++ )
    {
        switch( cmdBufIn[i] )
        {
            // STORE command/payload
                // Cases with 2 Byte payload
            // case XXX:
            // {
            //     cmdBufOut[cmdCount++] = cmdBufIn[i++];
            //     cmdBufOut[cmdCount++] = cmdBufIn[i++];
            //     cmdBufOut[cmdCount++] = cmdBufIn[i];
            //     break;
            // }
                // Cases with 1 Byte payload
            case MOTE_MAC_RX_PARAM_SETUP_ANS:
            {
                cmdBufOut[cmdCount++] = cmdBufIn[i++];
                cmdBufOut[cmdCount++] = cmdBufIn[i];
                break;
            }
                // Cases with  no payload
            case MOTE_MAC_RX_TIMING_SETUP_ANS:
            {
                cmdBufOut[cmdCount++] = cmdBufIn[i];
                break;
            }

            // DISCARD command/payload
                // Cases with 2 Byte payload
            case MOTE_MAC_DEV_STATUS_ANS:
            {
                i++;
                i++;
                break;
            }
                // Cases with 1 Byte payload
            case MOTE_MAC_LINK_ADR_ANS:
            {
                i++;
                break;
            }
                // Cases with  no payload
            case MOTE_MAC_LINK_CHECK_REQ:
            case MOTE_MAC_DUTY_CYCLE_ANS:
            case MOTE_MAC_NEW_CHANNEL_ANS:
            default:
                break;
        }
    }

    return cmdCount;
}
```

djaeckle, 2016-10-21:
Hi Mobiak,

thanks for the report. I will double check it and will come back to you soon.

djaeckle, 2016-10-27:
Hi Mobiak,

thanks again for the report. I've just pushed a fix on the develop branch.

closed 2016-10-27 by djaeckle