Lora-net/LoRaMac-node #153: MLME JOIN looping when no network respond.

mbracq, 2016-11-04:
Hi all,
I use the latest  version of the Loramac stack. 
When we do
```
  mlmeReq.Type = MLME_JOIN;  
  mlmeReq.Req.Join.DevEui = raub_swp_lorawan_DevEui;
  mlmeReq.Req.Join.AppEui = AppEui;
  mlmeReq.Req.Join.AppKey = AppKey;
  
  LoRaMacMlmeRequest( &mlmeReq );
```
and when no network responds, the stack sends request in loop.

We enter in _OnMacStateCheckTimerEvent_ function
with `LoRaMacFlags.Bits.MacDone = 1`
`MlmeConfirm.MlmeRequest = MLME_JOIN`
and `MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT`
And the stack schedule a TX with same data content immediately (line 1446). We didn't condition to stop this process . 

It's the exceped behavior?

Thanks in advance,
Manuel

rokobert, 2016-11-06:
Same problem here!
Bugy code.
I'm also look in specification and there is no explanation how to deal with problem when SF7 is to fast to join. This problem is partly solved in "LoRaMacMlmeRequest" where "JoinRequestTrials" variable is incremented and then SF is calculated again in "AlternateDatarate" function. If you look inside this function there is also strange calculation of SF. For example, if you need SF12 to join it will take 48 increments. Wery long period! But first of all, code must first get to there.
I will rewrite "AlternateDatarate" function. And add code to "OnMacStateCheckTimerEvent".

With code in bottom you will get event in "MlmeConfirm" calback  with Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT and then you could call again Join with "LoRaMacMlmeRequest" function.

        if( ( NodeAckRequested == false ) && ( txTimeout == false ) )
        {
            if( LoRaMacFlags.Bits.MlmeReq == 1 )
            {
                if( MlmeConfirm.MlmeRequest == MLME_JOIN )
                {
                     // Stop retransmission
                     ChannelsNbRepCounter = LoRaMacParams.ChannelsNbRep;
                     UpLinkCounter = 0;
                }
            }
       }

mluis1, 2016-11-07:
@mbracq 
The Join procedure is meant to be executed periodically up until the node joins the network.

@rokobert 
Could you explain what you mean by the "problem when SF7 is to fast to join"?

The develop branch does indeed solves the Join trials by recomputing the DevNonce as well as by using the most adapted datarate.

Concerning the datarate management the "strange way of computing the SF" as you say is because the node must comply with the specified dutycycle transmission. 
Please take a look at chapter 7 "Retransmissions back-off" of LoRaWAN 1.0.2 specification.

 |  |  | 
------------ | ---------------- | -----------------
Aggregated during the first hour following power-up or reset | T0<t<T0+1 | Transmit time < 36Sec
Aggregated during the next 10 hours | T0+1<t<T0+11 | Transmit time < 36Sec
After the first 11 hours, aggregated over 24h | T0+11+N<t<T0+35+N N>=0 | Transmit time < 8.7Sec per 24h

The datarate sequence used ensures that the specified dutycycle is respected in all conditions as well as it makes the node connect to the network at the fastest possible datarate. (Reduces the end device power consumption while performing the join procedure).

If you aren't careful, in case you change the Join procedure you may have the risk to no be compliant with the specification anymore.

mbracq, 2016-11-07:
Thanks @mluis1 and @rokobert  for yours answers.

I was working on master branch... I switch to the develop version. I see the datarate changing now.

I've just a question about the current implementation. If no server responds, the join will be infinite. Does we have to implement a max trial function?

mluis1, 2016-11-07:
No, the Join is meant do be done infinitely up until a server answers. One of the main goals of an end-device it to be connected to a server.

closed 2016-11-11 by mluis1

sdugua, 2016-11-14:
Hi, I would like to push this idea of a max trial value. I agree that one of the main goal is to be connected to a server but in a lot of case, the other goal is the low power or the ability to send an ack or a nack of the installation process to a installator. If we can set a max trial number, then at the end the MlmeConfirm could be called with a nok status and the application could inform the installator or decide to start a new set of trials... This seems very important for me. In addition today there is now way to stop the execution of the join procedure cleanly.
Sebastien.