Lora-net/LoRaMac-node #154: How to test class_b？

yanzi418, 2016-11-07:
How to test the node of class_b mode without the latest web server support? Is there a server test? We need to use the project class_b?
use feature/class_b code

mluis1, 2016-11-07:
I know that there are some Class B server implementations under development but, I guess that they will 
not be released before the LoRaWAN 1.1 specification is released and Class B specification is moved from experimental to released.

As the Class B specification is currently experimental it means that some changes that may break the retro-compatibility could happen in the future.

The current feature/class_b branch has been tested against 2 different server implementations. But, those implementations are not publicly available yet. (and I don't know when they will)

My advise would be that you try to contact the different server providers in order to know if they have already implemented a Class B server or if they have the intention to do so in the near future.

closed 2016-11-07 by yanzi418

adonislixi, 2017-02-27:
@mluis1 can u tell what server  that implement class_b already?

mluis1, 2017-02-27:
The server providers that I know to have a preliminary but functional classB implementation are:
Actility https://www.actility.com/
Orbiwise https://www.orbiwise.com/en/home 

Potentially, there are others but I'm not aware of them. You may try to contact the LoRa-Alliance to know if there are others.