Lora-net/LoRaMac-node #156: Maximum number of trials in JOIN

sdugua, 2016-11-14:
Hi, I would like to push the idea raised in a closed subject of a max trial function in the JOIN procedure. 
I agree that one of the main goals of an end-device is to be connected to a server but in a lot of case, the other goals are the low power feature or the ability to send an ack or a nack of the installation process to an installer. If we can set a max trial number, then at the end the MlmeConfirm callback could be called with a nok status and the application could inform the installer or decide to start a new set of trials... This is very important because, in addition, at that time there is now way to stop the execution of the join procedure cleanly. Thank you for your feedback. Sebastien.

kjendal, 2016-11-14:
Senet recommends to our partners that they implement an exponential backoff on the timing of the JoinRequest messages, which eventually reaches a periodicity no less than the longer of the application's natural transmit period or the duty-cycle limit specified in 1.0.2, section 7.

Depending on the region the exact algorithm may not be a simple exponential backoff.  In FCC style regions with unknown number of channels available, the device should not backoff until all possible channel banks have been attempted.

The reference code should be modified in such a way as to permit the configuration of a maximum number attempts, after which an error is returned to the application (on the end-device).

-Dave Kjendal

mluis1, 2016-11-15:
Just to clarify things. The develop branch already implements the backoff functionality for all supported bands.

If you would like to change the Join procedure functionality I would recommend that you issue a change request for this to the LoRa-Alliance technical committee. Currently the specification doesn't say anything about this.

The way it is currently implemented is to support for instance a sealed battery powered water meter that get out of factory and that is able to automatically connect to the network once installed at its final location.

sdugua, 2016-11-15:
Hi Miguel, I understand your point of view. However, I think it could be valuable to, at least, quickly add a way to stop the JOIN process from the application. This is not related to the lorawan specification for me but to a clean implementation. Do you think it can be pushed in the pipe ?

mluis1, 2016-11-15:
We will try to add this option in a future version. For the moment I can't say when it will happen.

What I propose is to add somewhere a parameter where:
0: Join is done indefinitely
X: Number of trials before reporting an error.

With this proposal everyone gets happy.

sdugua, 2016-11-15:
This is definitely the best solution.
Thank you a lot for your fast support on this.

closed 2017-01-16 by djaeckle

sda3422, 2017-01-24:
@djaeckle : Thank you for taking into account this need, however, I have few remarks.
The idea is to try to take into account on field installation by a human waiting for an installation status.
First, in the function **LoRaMacMlmeRequest** you force Req.Join.NbTrials = 48 minimum so, if I set this value to 3 it is overwritten (no really useful then).
Then, the **AlternateDatarate** is designed to work with this kind of high value. With a 10 seconds delay between the tries, the installation guy on field will wait a very long time if the unit is only accessible through DR0 (worst case of course).
It could be very valuable to be more flexible on the join procedure to take into account different way of working (low power, installation by a human waiting a feedback, ...).
What do you think ? Thanks.