Lora-net/LoRaMac-node #157: Join and Re joi

JulienArmenti, 2016-11-17:
Hi,

I have a question about this situation:

A node do an OTA activation with a gateway. After the activation, the node send some packet and everything is alright.

But, the gateway restarts (reset due to electrical alimentation shut down)

The node continu to send data but the gateawy don't receive them.

A solution exist to bypass this problem or is more efficiently to do a join procedure evey time I want send data?

Thanks
J.A

abouillot, 2016-11-17:
Julien,

The join procedure will require to exchange data between the node and the gateway. This will impact the air time available to the gateway to even accept the join request. Rejoin for every transmission is definitely a bad idea, as you might end up with the node not even capable to send its data, as not able to join.

It would depend on the network deployment. If several gateways are in radio distance, the message will be received by the other and be carried to your backend application. The node will not even notice a gateway went off. This is one of the main concept of a LoRaWAN network, with overlapping of the areas served by the gateways

You can potentially, a couple times in the day send a packet requesting acknowledge to feedback the node its data are received, but if no gateways are listening, there would be no fallback, apart to try rejoin periodically until the gateways are up again.

Alexandre

mluis1, 2016-11-25:
Can we close the issue?

JulienArmenti, 2016-11-25:
Yes, we can close this isssue.

Thanks a lot

J.A

From: Miguel Luis 
Sent: Friday, November 25, 2016 10:33 AM
To: Lora-net/LoRaMac-node 
Cc: JulienArmenti ; Author 
Subject: Re: [Lora-net/LoRaMac-node] Join and automatic rejoin (#157)

Can we close the issue?

closed 2016-11-25 by mluis1