Lora-net/LoRaMac-node #159: Maximum payload length setting problem at 433Mhz

ckt1010, 2016-11-22:
Hi,
Thanks to co-work with @adonislixi on issue #151, we found there is a setting problem in funtion RxWindowSetup:
Radio.SetMaxPayloadLength( modem, MaxPayloadOfDatarate[datarate] + LORA_MAC_FRMPAYLOAD_OVERHEAD );

As we use 433Mhz, so MaxPayloadOfDatarate with DR_5 will choice 250, and LORA_MAC_FRMPAYLOAD_OVERHEAD is 13. Then  maxpayloadlength will set as 263. But this function declare as below:
void ( *SetMaxPayloadLength )( RadioModems_t modem, uint8_t max );

This is why DR_5 packet can't receive but DR_0 can. And I notice 915Mhz and 868Mhz is use 242, do you think we should change it to 242 also? 
Thanks!

mluis1, 2016-11-22:
Thanks for the report. The issue comes from the specification itself. We took the values written in the document.

You are right we should change the value to 242.

For now I think that you can use the same table as for EU868.

I will ask the LoRa-Alliance technical committee to correct the maximum payload tables.

ckt1010, 2016-11-22:
Thanks @mluis1 . I check specification, still have a question can't understand:
[image]
[image]
The second one is EU433 maximum payload size.
Seems you use 'N' at EU902-928 but use 'M' at EU433. Since it already plus LORA_MAC_FRMPAYLOAD_OVERHEAD, shouldn't we use 'N' is more suitable?
Thanks!

mluis1, 2016-11-22:
Yes, you are right. We should use N column instead.

We will provide a fix as soon as possible.

ckt1010, 2016-11-22:
Thanks! Don't forget CN779-787, it also use 'M'.

mluis1, 2016-11-22:
Yes, we will review all tables for all bands. 

Thanks again for the provide input. The most tested bands are EU868 and US915 as they have LoRa-Alliance certification test already in place.

djaeckle, 2017-01-16:
Hi ckt1010,

we have updated the develop branch with the fixes.

closed 2017-01-16 by djaeckle