Lora-net/LoRaMac-node #160: disable_irq/enable_irq in TimerStart() and TimerStop()

denchii, 2016-11-22:
Dear developers,

I've got a quiestion - is it necessary to disable all interrupts in TimerStart() and TimerStop() functions? Is it possible to disable only RTC_Alarm_IRQn to prevent racing?
My goal is to catch every GPIO interrupt, but that mechanism leads to miss some pulses. 

`
void TimerStart( TimerEvent_t *obj )
{
    uint32_t elapsedTime = 0;
    uint32_t remainingTime = 0;

    __disable_irq( );

....
    __enable_irq( );
}
`

`
void TimerStop( TimerEvent_t *obj )
{
    __disable_irq( );
...
    __enable_irq( );
}
`

ckt1010, 2016-11-23:
I saw it has been changed on Develop branch, maybe you can check that.

mluis1, 2016-11-25:
Yes, the disable and enable IRQ are necessary in order to avoid having the interrupt handler changing the value while we are initializing it. We must consider the variables initialization as a critical section.

closed 2016-11-25 by mluis1