Lora-net/LoRaMac-node #162: RX information in McpsConfirm callback

sdugua, 2016-11-28:
Hello, I would like to recover the information of RSSI, SNR, RxSlot, etc on reception of an acknowledgment.
The McpsConfirm callback only takes in parameter the McpsConfirm structure that do not include these information (only in sMcpsIndication structure). Do you see a way to recover sMcpsIndication data in McpsConfirm callback or is it something missing today in the implementation ? Thank you !

mluis1, 2016-11-29:
Each time a downlink takes place the McpsIndication callback is called by the MAC layer.

In case of confirmed messages the McpsIndication is called and then the McpsConfirm callback is called.

The application examples for MoteII platform on develop branch shows how to handle the McpsIndication.

sdugua, 2016-11-29:
Hi Miguel, thanks a lot and sorry because it was my mistake !

closed 2016-11-29 by sdugua