Lora-net/LoRaMac-node #165: State of  LoRaWAN-lib-dev & SX1272 mbed libraries

loverdeg-ep, 2016-12-02:
Will these continue to be maintained?
LoRaWAN-lib-dev is pretty far behind.

Are there any plans to update the libraries and examples for mbed-os 5?
The mbed-cli tools use git by default now so if being forced to use mercurial is the issue, it might be preferable to maintain the libraries on github.

[https://developer.mbed.org/teams/Semtech/code/LoRaWAN-lib-dev/](https://developer.mbed.org/teams/Semtech/code/LoRaWAN-lib-dev/)
[https://developer.mbed.org/teams/Semtech/code/SX1272Lib/](https://developer.mbed.org/teams/Semtech/code/SX1272Lib/)

mluis1, 2016-12-02:
I'm currently working on synchronizing the libraries on mbed (I do it when I have some spare time.)
It should be available in the coming two weeks.

The libraries to be updated are:
https://developer.mbed.org/teams/Semtech/code/LoRaWAN-lib-dev/
https://developer.mbed.org/teams/Semtech/code/SX1272Lib/
https://developer.mbed.org/teams/Semtech/code/SX1276Lib/

The mbed-os 5 support will require some more work.
Up until now I didn't had the time to look at it and it is not yet planned.

loverdeg-ep, 2017-03-24:
Regarding the future inclusion of lorawan to mbed os 5.

https://github.com/ARMmbed/mbed-os/blob/feature-lorawan/features/FEATURE_LORAWAN/todo.md

Considering it looks like their plan is to wrap up the source already hosted [here](https://github.com/Lora-net/LoRaMac-node), will the mbed "C++" version" of LoRaMac-node and radio drivers continue to be maintained?

mluis1, 2017-04-24:
All LoRaWAN projects on mbed Semtech team page have been synchronized with this repository master branch.

closed 2017-04-24 by mluis1

loverdeg-ep, 2017-05-19:
Were these updates tested?
My gateway isn't seeing any packets from a Namote72 loaded with the https://developer.mbed.org/teams/Semtech/code/LoRaWAN-NAMote72-Application-Demo/

Other than triple checking the configuration I haven't had the time to dig too deep yet.

snelson-senet, 2017-05-19:
I also see  this issue.   It does work if I change the project revision to the one just prior to the github sync.

maxmclau, 2017-11-15:
Do we know if anyone is currently working on the FEATURE_LORAWAN integration? Would we be waiting on Semtech for that?