Lora-net/LoRaMac-node #166: Class C downlink issue.

batulzii, 2016-12-05:
Using sx1276 lib for modified LoraMote. uplink is ok. downlink is good if haven't uplink before. Any uplink will stopget downlink any more.  For Class C mode.

batulzii, 2016-12-06:
A quick fix would be to comment out Radio.Standby(); in the OnRxWindow1TimerEvent

if (LoRaMacDeviceClass == CLASS_C) {
//Radio.Standby();
}

But need to check why put radio to standby before Window2?

Pupil2013, 2016-12-07:
we can see that in the function:it stops the RXtimeroutTimer,that's right according to the specification continuous listening no rxtimeout.it will set the radio in receieve mode later in function RxWindowSetup();
void SX1276SetStby( void )
{
    TimerStop( &RxTimeoutTimer );
    TimerStop( &TxTimeoutTimer );

    SX1276SetOpMode( RF_OPMODE_STANDBY );
    SX1276.Settings.State = RF_IDLE;
}

Pupil2013, 2016-12-07:
Before we change the chip's mode ,we need to set it to standby mode.
That's why we need to put radio to standby before Window2;

batulzii, 2016-12-07:
RxWindowSetup( Channels[Channel].Frequency, datarate, bandwidth, symbTimeout, false );
makes rxmode for A /B only. seems to be add case for class C with 
    RxWindowSetup( Channels[Channel].Frequency, datarate, bandwidth, symbTimeout, true);
?

Pupil2013, 2016-12-08:
makes rxmode for A /B only. ---------that's not true.It's the same for CLASS C

batulzii, 2016-12-13:
only works when comment in the OnRxWindow1TimerEvent

if (LoRaMacDeviceClass == CLASS_C) {
//Radio.Standby();
}
 otherwise uplink could not received. (Class C). Is It the same bug source as JOIN ACK problem

Pupil2013, 2016-12-15:
It maybe the time issue.The node may open the recieve slot2 in wrong time or there eixts some time error.
"The receive windows should  achieve 5 overlapping symbols with the downlink preamble"
when you comment the standby() function,the node will stay in receive mode.So the receive time error don't exits any more. @batulzii

batulzii, 2016-12-15:
Thanks for kind support! Please, advice how to debug time issue.

Pupil2013, 2016-12-15:
`if( ( LoRaMacParams.Rx2Channel.Datarate == DR_3 ) || ( LoRaMacParams.Rx2Channel.Datarate == DR_4 ) )
    { // DR_4, DR_3
        symbTimeout = 8;
    }
    else if( LoRaMacParams.Rx2Channel.Datarate == DR_5 )
    {
        symbTimeout = 10;
    }`
you can try to increase the symbTimeout value. Make the receive slot longer.

batulzii, 2016-12-15:
I tested with several increase without success. Interesting point that downlink is ok after reset. but nomore downlink, if sent uplink  at least once.

mluis1, 2017-02-24:
Can we close the issue?

closed 2017-02-24 by batulzii