Lora-net/LoRaMac-node #167: enable the PA_BOOST to have +20dBm

FoujanetG, 2016-12-06:
Hi,

I would like to enable the PA_Boost to have +20dBm in the radio output with 915MHz frequency band:

I think we need to play with 3 registers:
**paConfig =0xFF** 
 _PaSelect_(7) =1 -->Pa_Boost selected
 _MaxPower_(6-4)=111 --> we don't care, only useful for 14dBm
 _Output Power_(3-0) = 1111 -->Pout= 17 - (15 - 15)=17dBm(PA_BOOST PIN is selected) 

Why Output Power is only 17dBm? How to get 20dBm?

**paDac = 0x87**
_paDac_ (2-0) =0x7 -->  +20dBm on PA_BOOST when OutputPower=1111

**REG_OCP:over current protection**==> 
By default, the current in output is limited to 100mA: the currently stack doesn't play with this register
With Pa_Boost at +20dBm, I need 120mA of current.(see datasheet sx127x ,chapter _2.5.1. Power Consumption_)
So I changed the over current protection, and I set the register to have  140mA .

In output of the radio, I measure +17dBm. 
Is there a problem in my reflexion to have +20dBm?

thanks

Pupil2013, 2016-12-07:
Also has the same confuse.According to the datasheet ,there is no way to output 20dBm

dsMartyn, 2017-01-13:
have you measured the output power on PA_Boost? 
i have measured 20dBm into the RF path when using a 50Ω dummy load, did you measure the TX current to see if you are getting the +120mA

mluis1, 2017-01-24:
@FoujanetG 
It is not very clear which radio you are trying to set. The 20 dBm output isn't handled the same way in sx1272 and sx1276.
From what I understand in your question I think you are using the sx1276 radio.

The radio drivers provided in example here handle correctly the PA_BOOST as well as the 20 dBm output.
Please take a look at the below function in sx1276-board.c file.

Please also note that the formula to compute the power register value changes when using 20 dBm mode.

When the 20 dBm option is enabled the power current protection is automatically managed by the radio hardware. Thus there is no need to change the register value.

Another thing that need to be taken care is the antenna matching at 915 MHz which could affect your output power if matched for 868 MHz for instance.

```C
void SX1276SetRfTxPower( int8_t power )
{
    uint8_t paConfig = 0;
    uint8_t paDac = 0;

    paConfig = SX1276Read( REG_PACONFIG );
    paDac = SX1276Read( REG_PADAC );

    paConfig = ( paConfig & RF_PACONFIG_PASELECT_MASK ) | SX1276GetPaSelect( SX1276.Settings.Channel );
    paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70;

    if( ( paConfig & RF_PACONFIG_PASELECT_PABOOST ) == RF_PACONFIG_PASELECT_PABOOST )
    {
        if( power > 17 )
        {
            paDac = ( paDac & RF_PADAC_20DBM_MASK ) | RF_PADAC_20DBM_ON;
        }
        else
        {
            paDac = ( paDac & RF_PADAC_20DBM_MASK ) | RF_PADAC_20DBM_OFF;
        }
        if( ( paDac & RF_PADAC_20DBM_ON ) == RF_PADAC_20DBM_ON )
        {
            if( power < 5 )
            {
                power = 5;
            }
            if( power > 20 )
            {
                power = 20;
            }
            paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 5 ) & 0x0F );
        }
        else
        {
            if( power < 2 )
            {
                power = 2;
            }
            if( power > 17 )
            {
                power = 17;
            }
            paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 2 ) & 0x0F );
        }
    }
    else
    {
        if( power < -1 )
        {
            power = -1;
        }
        if( power > 14 )
        {
            power = 14;
        }
        paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power + 1 ) & 0x0F );
    }
    SX1276Write( REG_PACONFIG, paConfig );
    SX1276Write( REG_PADAC, paDac );
}
```

@Pupil2013 
The datasheet provides all required information on how to set the radio to transmit at 20 dBm.
Please read chapter "5.4.3. High Power +20 dBm Operation" of SX1276 datasheet.

closed 2017-02-03 by mluis1