Lora-net/LoRaMac-node #168: Miss ISR variables protection

NICOLASGON, 2016-12-06:
Hi,

Dio ISR functions are not ISR safe, there is shared variables with main program but you don't deactivate the ISRs.

Sincerely yours,
ngonzalez

mluis1, 2016-12-12:
We are aware of this situation and we don't consider it as problem for the moment. 
We keep this in mind when doing the code review process.

matijsbrs, 2016-12-16:
Hi,

I'm currently having lots of downlink issues with Class C devices. They just stop receiving. 
The downlink starts working again after either a reinitialisation of the stack or simply by sending a frame in the uplink. 

The states in the 'SX1272.Settings.LoRa' struct are exactly what  one would expect. 

This problem occurs randomly, but evidence is more and more pointing in the direction of issues with the interrupts.

GuntherScuhlz, 2017-01-12:
Hi @matijsbrs 

I had similar problem to you with Class A device when receiving **duplicate** confirmed downstream frames. Using V4.3 code base.

Finally traced the error to OnRadioRxDone() in LoRaMac.c, Line 1179.

What happened in my case was, when duplicate confirmed downstream frame happens, this offending line has the net effect of silently ignoring the rx event (by NOT setting LoRaMacFlags.Bits.McpsInd = 1 in line 1283). Therefore OnMacStateCheckTimerEvent never gets to clear MAC_TX_RUNNING flag at line 1463, so we never enter LoRaMacState == MAC_IDLE at line 1518, so McpsConfirm never gets called in line 1522, nor does LoRaMacFlags.Bits.MacDone get reset at line 1532, nor does McpsIndication get called at line 1543.

This has the net effect that although your radio hardware has done all it needs to, the Mac layer still hangs around forever thinking it is waiting for data (that will never come, since it has been suppressed, in associated timeouts killed in radio layer). 

Sending an uplink frame restartss a new recieve window with its own new timeouts etc, which silently cleans up this incorrect state when IT finishes.

Resetting the stack obviously has the same effect.

I patched my system as follows, and never had the issue again

Add a global                    "bool DuplicateRxPacket = false;"

Line   933: Add                "DuplicateRxPacket = false;"
Line 1179: Add                "DuplicateRxPacket = true;"
Line 1281: CHANGE to    "if(( skipIndication == false ) || (DuplicateRxPacket == true))"

Then in your McpsIndication function, explicitly check DuplicateRxPacket on entry, and return immediatlely if it is set.

Its a nasty kludge, but its minimal, and works flawlessly, 

Hope this helps bro :)

GuntherScuhlz, 2017-01-12:
@djaeckle @mluis1 @reissjason @sbress @nestorayuso @sylvain-miermont 

Hi contirbuters, please check my findings on #168  above

Thanks

GuntherScuhlz, 2017-01-12:
related to #122 and #125

djaeckle, 2017-01-12:
Hi  GuntherScuhlz,

thanks for your report. On which branch/commit are you working on?

GuntherScuhlz, 2017-01-13:
Hi Daniel

Only a pleasure :)

Master branch,

2016-06-22, V4.3.0

Regards,

Gunther Schulz

<comment@noreply.github.com>

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com/email-signature>
Version: 2016.0.7996 / Virus Database: 4749/13750 - Release Date: 01/11/17

djaeckle, 2017-01-13:
Hi GuntherScuhlz,

does the issue also occur when you are using the develop branch?

GuntherScuhlz, 2017-01-13:
Hi Daniel

I never bothered to look at develop branch, as I didn’t have time, and my 
solution worked for me.

Regards,

Gunther Schulz

Version: 2016.0.7996 / Virus Database: 4749/13756 - Release Date: 01/12/17

mluis1, 2017-01-26:
Can we close this issue?

GuntherScuhlz, 2017-01-26:
Sure

Version: 2016.0.7998 / Virus Database: 4749/13838 - Release Date: 01/26/17

closed 2017-01-26 by mluis1