Lora-net/LoRaMac-node #170: error in Keil Project Build.

twowinter, 2016-12-09:
error: #268: declaration may not appear after executable statement in block

The Standard C don't allow the code like this:
`    uint32_t elapsedTime = 0;`

Is CoIDE recommended?

twowinter, 2016-12-12:
`    uint32_t elapsedTime;

    elapsedTime = 0;`

Keil MDK would build success while change code to this,.

mluis1, 2016-12-12:
Which Keil project are you talking about?

On which file and line the compiler makes the error?

I'm wondering if you haven't changed the project settings. The declaration of variables in the middle of the code is allowed by the C99 version of the C specification. All projects examples have this option enabled.

If you remove the C99 option you will have the same issue with all compilers/development environments.

twowinter, 2016-12-12:
Thank you!

As you metion C99, i find out the reason.

MDK(version 4.7.0) are no C99 defined default, after i add --C99, it be OK.

[image]

closed 2016-12-12 by twowinter