Lora-net/LoRaMac-node #171: Set default channel from 919Mhz to 928Mhz

fwndz, 2016-12-11:
Hi,

I have question regarding setting up the default channel for 915 Mhz bandwidth. If I want to set the transmission frequency only in between 919 MHz and 928 MHz what configuration should I add in to the code?  I saw in main.c there are some configuration had been done for 868 MHz as shown below.
```
...
#define USE_SEMTECH_DEFAULT_CHANNEL_LINEUP          1

#if( USE_SEMTECH_DEFAULT_CHANNEL_LINEUP == 1 ) 

#define LC4                { 867100000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC5                { 867300000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC6                { 867500000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC7                { 867700000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC8                { 867900000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC9                { 868800000, { ( ( DR_7 << 4 ) | DR_7 ) }, 2 }

#endif

#endif

...
#if defined( USE_BAND_868 )
                LoRaMacTestSetDutyCycleOn( LORAWAN_DUTYCYCLE_ON );

#if( USE_SEMTECH_DEFAULT_CHANNEL_LINEUP == 1 ) 
                LoRaMacChannelAdd( 3, ( ChannelParams_t )LC4 );
                LoRaMacChannelAdd( 4, ( ChannelParams_t )LC5 );
                LoRaMacChannelAdd( 5, ( ChannelParams_t )LC6 );
                LoRaMacChannelAdd( 6, ( ChannelParams_t )LC7 );
                LoRaMacChannelAdd( 7, ( ChannelParams_t )LC8 );
                LoRaMacChannelAdd( 8, ( ChannelParams_t )LC9 );
#endif

#endif
```
So for my case, could you give me some guideline to setting up the LoraMote Channel which can transmit only above 919 MHz since this is the requirement in our country. 

Thanks 

Regards
Fndz

djaeckle, 2017-01-13:
Hi fwndz,

in which country do you want to use it?

mluis1, 2017-02-03:
Can we close this issue?

closed 2017-02-09 by mluis1