Lora-net/LoRaMac-node #172: Class C OTAA - LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT

batulzii, 2016-12-13:
my loramote is on class c. works ok with ABP, but OTAA is without ACK.

2016-12-13 22:43:02.616	JOIN_ACCEPT	Join accept sent to deveui=BE7A00000000031A via gw=00-23-CD-FF-FF-00-0E-7E.0 freq=868.500 datr=SF12BW125
2016-12-13 22:43:02.611	JOIN_REQUEST	Join request to appeui=1122334455667711 from DEVEUI

log on loriot.

debuged received  LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT  on node side.

edited as bellow:

#define LORAMAC_PHY_MAXPAYLOAD                      242

but still could not get ACK.

changed datarate of Join to SF9
but still could not get ACK.

djaeckle, 2016-12-20:
Hi batulzii,

thanks for the report. In issue #166, you mentioned that you have a modified LoRaMote. Could you provide some information about the modifications? Which LoRaMac-node version are you using?

If the node reports a `LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT`, the node was not able to handle the frame transmission properly.

batulzii, 2016-12-20:
Loramote based on HopeRF module and L70 GPS and some minor adds. It works fine with ABP. Seems to be issue with Class C. Transmission done if check logs. Thanks for kind support!

djaeckle, 2017-02-09:
Hi batulzii,

could you please provide the changes?

mluis1, 2017-02-24:
Can we close the issue?

closed 2017-02-24 by batulzii