Lora-net/LoRaMac-node #173: Functions in ISR context

jia200x, 2016-12-15:
Hi.

In our port of LoRaMAC some functions like OnMacStateCheckTimerEvent are running in ISR context due to timer callback. It seems the original implementation does the same. If this is correct, an ISR flags based solution might be better.

When we try Over The Air Activation with Atmel SAMR21-xpro, it crashes due to ISR stack overflow. This doesn't happen with STM32L1 based MCUs.

Cheers

mluis1, 2017-01-24:
Duplicates Issue #168

jia200x, 2017-01-24:
Hi @mluis1 .
I think is not the same issue. The problem is the ISR stack overflow due to big chunks of code in ISR context. It's not fixed with sync of variables.
Cheers

mluis1, 2017-01-25:
Ah, ok my mistake, I will remove the duplication comment.

mluis1, 2017-01-25:
Personally, I have no experience with the SAMR21-xpro mcu. 

But, looking at its characteristics I don't see an obvious reason why an overflow happens.

The MCU uses the same ARM architecture as the examples provide on this project. In plus this MCU has bigger RAM and FLASH memories than the ones used in the examples provided here.

Another thing, using OTAA or ABP activation should have no influence in the way the stack behaves as both methods are manged the same way by the MCU.

fjmolinas, 2017-01-26:
We have been having the same issue. The timer callback goes to OnMacStateCheckTimerEvent, witch executes code in ISR context. Depending on specific conditions it will end up calling for the ScheduleTx(), witch SendFrameOnChannel(), witch calls another bunch of functions. Since OnMacStateChecTimerEvent hasn't been exited yet all this is happening in ISR context, witch is too much code and can exhaust the ISR stack for certain boards.

GuntherScuhlz, 2017-04-10:
I happen to be blessed with enough RAM to run the stack as-is, but i definately +1 the fact that spending oodles of time in ISR context is evil, this pattern needs to change. Perhaps just set flags in ISRs, and have the external loop react to them and call processing chain as normal, or defer ISRs to some user-defined low priority non-blocking check-and-action scheme to handle this.

GuntherScuhlz, 2017-06-21:
Hi @mluis1 , just bumping the topic. I see you did tag this as a task - does this imply you're going to look into it after 4.4.0 release?

mluis1, 2017-06-21:
Yes, as we are close to finalize the 4.4.0 release (under certification process) this task will be applied to 5.0.0 release development.

mikkleini, 2017-06-27:
I have some doubts about the thread safety also. LoRa state machine is executed in the main loop (or in some thread) but suspiciously lot of stuff is happening in ISR's. There are no critical sections, locks, mutexes. Is it really guaranteed that when interrupt preempts the main loop then nothing goes out of sync?
I don't say there is an issue, but based on experience such solutions are error prone and can cause rare and hard to find issues.
+1 to the pattern which Gunther suggests - ISR's just raise flags (in critical section) and main loop acts on them and clears them (in protected section). Or if possible, don't use ISR's at all - poll the interrupt flag directly from main loop.

closed 2018-07-02 by mluis1