Lora-net/LoRaMac-node #175: About CAD mode

adonislixi, 2016-12-21:
In datasheet , CAD mode can reduce the reception power consumption.

In Master/develop , I  see RadioEvents->CadDone in the Dio3 interrupt, but the pointer is not assigned.

anyone has the experience about how to use CAD mode？

@mluis1 @djaeckle

mluis1, 2017-01-24:
The CAD feature is only useful to detect LoRa preamble symbols (excludes other modulations types and LoRa data symbols). The CAD operation usually requires the transmitter to transmit long preamble sequences.

Most of the time this feature is used when implementing a point to point protocol where the transmitter is main powered and the receiver does an duty-cycled reception.

For the LoRaWAN use case there is no need to use the CAD feature as we know in advance when the reception windows happen. Usually 1 and 2 seconds after the end of the transmission.

closed 2017-02-03 by mluis1

oyj0594, 2017-09-22:
@mluis1 
Hi, you mean that the CAD feature just sense LoRa preamble?
so, CAD could not sense LoRa data symbols or other signals in the same frequency area?