Lora-net/LoRaMac-node #176: HAL I2c problems with iM880b module

MvanEerd, 2016-12-22:
Hello,

I'm currently working on a project with a iM880b module. This is a newer version of the iM880a module which is used in the SK-iM880a and Loramote projects. 

I've made my own PCB and have been testing it with an older version of this project which uses the Std_Peripheral Librairy. One of the things I have connected to the iM880b module is a mpl3115 pressure sensor which is also used by the Loramote. With the old software version this worked like it should be, but in the new version which uses the HAL driver v1.1.3 the I2c stopped working correctly. 

I've checked the I2c bus with a scope and checked the bits which are send over the i2c bus with a decoder on the scope. It seems like the Ack which should be received isn't received by the software. Could someone confirm if they got the mpl3115 working with the Loramote by using the HAL_driver? 

Because the Ack isn't received I only see the I2c adress passing on the scope. The error "HAL_I2C_ERROR_AF" I got from the function HAL_I2C_Mem_Write() is generated in the stm32l1xx_hal_i2c.c file.

Other suggestions are also welcome. 

Thanks in advance,

MvanEerd

pdemil, 2016-12-22:
Check your system clock frequency and your I2C clock frequency  (100 kHz, 400 kHz?)

MvanEerd, 2016-12-22:
For the I2c clock frequency I've tried both the 100kHz and 400 kHz. On the scope I can verify that the frequenty is changed because of the wider clock pulses.(4 times wider) 

I've not changed the system clock frequency so this is still standard.

pdemil, 2016-12-22:
So you have also Checked your I2C pin configuration?

MvanEerd, 2016-12-22:
I've created a basic I2c program based on the HAL_Driver which is used in the LoRaMac-node software. It creates the same problem as in the LoRaMac-node software:

[image]

 I've used the same I2c pins as the Loramote. So it should work on a Loramote if someone has one. The code which I used to create the output as shown above:

```
#include "stm32l1xx_hal.h"

I2C_HandleTypeDef hi2c1;

void timer_Init(void);
void I2C_Init(void);

int main (void)
{
	HAL_Init( );
	HAL_MspInit();
	
	timer_Init();
	
	__HAL_RCC_GPIOA_CLK_ENABLE( );

	// Init Buzzer
	GPIO_InitTypeDef GPIO_InitStruct_Buzzer;
	GPIO_InitStruct_Buzzer.Pin       = GPIO_PIN_8;
	GPIO_InitStruct_Buzzer.Mode      = GPIO_MODE_OUTPUT_PP;
	GPIO_InitStruct_Buzzer.Pull      = GPIO_NOPULL;
	GPIO_InitStruct_Buzzer.Speed     = GPIO_SPEED_HIGH;
	HAL_GPIO_Init(GPIOA, &GPIO_InitStruct_Buzzer);

	// Init Enable Pin (need to be Low)
	GPIO_InitTypeDef GPIO_InitStruct_Enable;
	GPIO_InitStruct_Enable.Pin       = GPIO_PIN_11;
	GPIO_InitStruct_Enable.Mode      = GPIO_MODE_OUTPUT_PP;
	GPIO_InitStruct_Enable.Pull      = GPIO_NOPULL;
	GPIO_InitStruct_Enable.Speed     = GPIO_SPEED_HIGH;
	HAL_GPIO_Init(GPIOA, &GPIO_InitStruct_Enable);

	HAL_GPIO_WritePin(GPIOA, GPIO_PIN_11, 0);

	I2C_Init();

	uint8_t dummieBuf[2];
	dummieBuf[0] = 0x26;
	dummieBuf[1] = 0x01;

	while (1) {
		HAL_GPIO_WritePin(GPIOA, GPIO_PIN_8, 1);
		HAL_Delay(1000);
		HAL_GPIO_WritePin(GPIOA, GPIO_PIN_8, 0);
		HAL_Delay(20000);

		HAL_I2C_Master_Transmit(&hi2c1,(0x60<<1), (uint8_t*)dummieBuf, 0x02,5000);

		HAL_I2C_Master_Transmit(&hi2c1,(0x60<<1), (uint8_t*)dummieBuf, 0x02,5000);
	}
}

void timer_Init(void)
{
	RCC_OscInitTypeDef RCC_OscInitStruct;
	RCC_ClkInitTypeDef RCC_ClkInitStruct;
	RCC_PeriphCLKInitTypeDef PeriphClkInit;

	__HAL_RCC_PWR_CLK_ENABLE( );

	__HAL_PWR_VOLTAGESCALING_CONFIG( PWR_REGULATOR_VOLTAGE_SCALE1 );

	RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE|RCC_OSCILLATORTYPE_LSE;
	RCC_OscInitStruct.HSEState = RCC_HSE_ON;
	RCC_OscInitStruct.LSEState = RCC_LSE_ON;
	RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
	RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
	RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL6;
	RCC_OscInitStruct.PLL.PLLDIV = RCC_PLL_DIV3;
	HAL_RCC_OscConfig( &RCC_OscInitStruct );

	RCC_ClkInitStruct.ClockType = ( RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2 );
	RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
	RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
	RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
	RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
	HAL_RCC_ClockConfig( &RCC_ClkInitStruct, FLASH_LATENCY_1 );

	PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_RTC;
	PeriphClkInit.RTCClockSelection = RCC_RTCCLKSOURCE_LSE;
	HAL_RCCEx_PeriphCLKConfig( &PeriphClkInit );

	HAL_SYSTICK_Config( HAL_RCC_GetHCLKFreq( ) / 1000 );

	HAL_SYSTICK_CLKSourceConfig( SYSTICK_CLKSOURCE_HCLK );

	/*    HAL_NVIC_GetPriorityGrouping*/
	HAL_NVIC_SetPriorityGrouping( NVIC_PRIORITYGROUP_4 );

	/* SysTick_IRQn interrupt configuration */
	HAL_NVIC_SetPriority( SysTick_IRQn, 0, 0 );
}

void I2C_Init(void)
{
	GPIO_InitTypeDef GPIO_InitStruct;
	__HAL_RCC_I2C1_CLK_ENABLE();
	__GPIOB_CLK_ENABLE();
	/**I2C1 GPIO Configuration
		PB8     ------> I2C1_SCL
		PB9     ------> I2C1_SDA
	*/
	GPIO_InitStruct.Pin = GPIO_PIN_8|GPIO_PIN_9;
	GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
	GPIO_InitStruct.Pull = GPIO_PULLUP;
	GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
	GPIO_InitStruct.Alternate = GPIO_AF4_I2C1;
	HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

	hi2c1.Instance = I2C1;
	hi2c1.Init.ClockSpeed = 100000;
	hi2c1.Init.DutyCycle = I2C_DUTYCYCLE_2;
	hi2c1.Init.OwnAddress1 = 0;
	hi2c1.Init.AddressingMode = I2C_ADDRESSINGMODE_7BIT;
	hi2c1.Init.DualAddressMode = I2C_DUALADDRESS_DISABLE;
	hi2c1.Init.OwnAddress2 = 0;
	hi2c1.Init.GeneralCallMode = I2C_GENERALCALL_DISABLE;
	hi2c1.Init.NoStretchMode = I2C_NOSTRETCH_ENABLE;
	HAL_I2C_Init(&hi2c1);
	HAL_I2C_MspInit(&hi2c1);
}
```

x893, 2016-12-22:
Add correct I2C initialization. Because after reset slave device stay in unknown state.

MvanEerd, 2016-12-22:
Could you be more specific because I don't know which part of the initialization you refer to. The HAL driver won't send the RegAddr and data, so I can't reach the registers from the mpl3115. 

You are correct about the unknown state if the regAddr is send. But the HAL_driver expect an Ack of each I2c frame. 
For example to write to an register:                                
1. An Ack after sending I2c Adress    (0x60);                                   
2. An Ack after sending CTRL_REG1  (0x26) ;                  
3. An Ack after sending new value of CTRL_REG1;          

The example code cannot past step 1...

batulzii, 2016-12-22:
Have you change pins on I2C port. Check     I2C_1 = ( uint32_t )I2C1_BASE,
    I2C_2 = ( uint32_t )I2C2_BASE, ---- if change I2C port on your new schematic. So Accordingly change I2C_1 or I2C_2. Hope helped?

MvanEerd, 2016-12-23:
The i2c with the std_peripheral lib looks like this:

[image]

The i2c with the HAL_Driver looks like this:

[image]

It looks like the HAL hasn't defined something. I already tried to change the pins to PUSH_PULL, OPEN_DRAIN, NO_PULL, PULL_DOWN and PULL_UP with just a minor differences. I've created a new project using STM32CubeMX which create the same problem. If someone is interrested I could upload the project to a Git repository.

mluis1, 2017-01-24:
The issue comes from the fact that when you send the RESET I2C command to the MPL3115 device this one doesn't generate the ACK. The new HAL library looks to handle this situation differently from the former Std peripheral library.

The driver example provide in this project copes with this situation by instructing the MCU to reset its internal I2C peripheral after issuing the MPL3115 reset command.

```C
uint8_t MPL3115Init( void )
{
    uint8_t regVal = 0;

    MPL3115SetDeviceAddr( MPL3115A_I2C_ADDRESS );

    if( MPL3115Initialized == false )
    {
        MPL3115Write( CTRL_REG1, RST );
        DelayMs( 50 );
        I2cResetBus( &I2c ); // MCU I2C peripheral is rest here!!!!!

        // Check MPL3115 ID
        MPL3115Read( MPL3115_ID, &regVal );
        if( regVal != 0xC4 )
        {
            return FAIL;
        }

        MPL3115Write( PT_DATA_CFG_REG, DREM | PDEFE | TDEFE );      // Enable data ready flags for pressure and temperature )
        MPL3115Write( CTRL_REG1, ALT | OS_32 | SBYB );              // Set sensor to active state with oversampling ratio 128 (512 ms between samples)
        MPL3115Initialized = true;
    }
    return SUCCESS;
}

```

closed 2017-02-03 by mluis1