Lora-net/LoRaMac-node #178: The node cannot receive the ACK sometimes

eydonmlau, 2017-01-04:
Hi, I have a question. 
The node cannot receive the ACK sometimes when it  sends a confirmed data message to Gateway nearby. I find the RSS and SNR values(uplink) are satisfactory. There are several tests are conducted, and each test includes 10,000 data messages. The retransmission rate is 6%-15%. I may sure that the Gateway can receive the ACK provided by network server, but the node cannot receive the ACK. When I Debug  the node ,I found it  always go to OnRadioRxTimeout function  when retransmission  happen .
 Is it caused by  RxWindowDelay or something else ? What can I do to solve this problem?

ckt1010, 2017-01-06:
Hi,
I'm also run into this problem. My retransmission rate even higher than you. Which ISM band did you use?

eydonmlau, 2017-01-06:
hi ,ISM 470 ,
I will text on ISM868 and then offer the retransmission rate for you next week

ckt1010, 2017-01-14:
Hi @eydonmlau ,
Are you Chinese also?
Which LoRa-server did you use?
Now I suspect LoRa-server, or end-point Rx window length problem. Let's share our new finding.

djaeckle, 2017-01-16:
Hi,

we found an issue in the implementation for CN470. The symbol timeout for RX window 1 was not calculated correctly. This could possibly lead to receive problems in RX window 1.

ckt1010, 2017-01-17:
Hi @djaeckle ,
It is good news!
Do you have plan about when to fix this issue? I hope can fix it ASAP. :)
Thanks!

ckt1010, 2017-01-17:
Hi @djaeckle 
It's a good news!
But I see your commit is only change datarate variable name. Is that final fix? 
Thanks!

djaeckle, 2017-01-17:
Hi ckt1010,

correct. For RX window 1, the calculation of the symbol timeout was based on the wrong variable. It was `LoRaMacParams.Rx2Channel.Datarate`, which is the datarate for the RX window 2. For RX window 1, we have to use the variable `datarate`. Otherwise, we may set a symbol timeout which doesn't fit to the datarate. This may lead to RX problems in window 1.

ckt1010, 2017-01-23:
Hi @djaeckle ,
Unfortunately, I test your correction code again, but it is useless for our node. Would you please tell me your packet loss rate on CN470 ISM band?
Thanks!

djaeckle, 2017-02-09:
Hi ckt1010,

it's in progess. We try to provide the information as soon as possible.

mluis1, 2017-02-27:
Unfortunately, as of today we have no CN470 band setup in place.

The first thing that should be checked is if the reception windows are opened at the right time.
In order to be sure it is the case I would propose that you add the following debug pins to the radio driver SetOpMode function shown below.

Once the debug pins are in place you should observe something like the following oscilloscope screenshot. (a JoinReq is shown that's why the delays are 5 and 6 seconds).
In this case the reception window is opened 55 ms too earlier
[image]
If we zoom around 5 seconds we can better observe why the JoinAcept isn't received in this case.
The reception window is not opened at the right time.
[image]
 
```C
void SX1276SetOpMode( uint8_t opMode )
{
#if defined( USE_RADIO_DEBUG ) && defined( USE_DEBUG_PINS )
    switch( opMode )
    {
        case RF_OPMODE_TRANSMITTER:
            GpioWrite( &DbgPin1, 1 );
            break;
        case RF_OPMODE_RECEIVER:
        case RFLR_OPMODE_RECEIVER_SINGLE:
            GpioWrite( &DbgPin2, 1 );
            break;
        default:
            GpioWrite( &DbgPin1, 0 );
            GpioWrite( &DbgPin2, 0 );
            break;
    }
#endif
    if( opMode == RF_OPMODE_SLEEP )
    {
        SX1276SetAntSwLowPower( true );
    }
    else
    {
        SX1276SetAntSwLowPower( false );
        SX1276SetAntSw( opMode );
    }
    SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode );
}
```

mluis1, 2017-02-27:
My Chinese colleagues have run some tests on band 470 MHz.
The test has been performed on their office environment (Not conducted).
They have performed the test with one channel (470.3MHz) twice in a day.

The 1st time: end node sent out 460 packets, 5 of them retrained, gateway and server received 465 packets totally. So the re-transmission ratio is 1.09%

The 2nd time: end node sent 1090 packets, 8 of them retrained, gateway and server received 1098 packets totally. So the re-transmission ratio is 0.73%

The LoRaWAN certification tests defines a maximum PER of 5%. So in this case we are much lower than this  limit.

The only advice I can do is to check if your timing base is accurate.

mluis1, 2017-03-06:
Can we close the issue?

ckt1010, 2017-03-08:
Thanks @mluis1 , I'll check my code.

closed 2017-03-17 by mluis1