Lora-net/LoRaMac-node #179: Should we increase UpLinkCounter after maximum number of retransmissions?

ckt1010, 2017-01-05:
Hi,
I think it is a unclear point on LoRaWAN-spec 1.0:
From Chapter 4.3.1.3 
Note: If an end-device has reached its maximum number of retransmissions without receiving an  acknowledgment, it can try to regain connectivity by moving to a lower data rate with longer reach. It is up to the end-device to retransmit the message again or to **forfeit that message and move on.**

From this code we think this "forfeit that message and move on" is increase UpLinkCounter after maximum number of retransmissions. But from Chapter 4.3.1.5
If this difference is greater than the value of MAX_FCNY_GAP then too many data frames have been lost then subsequent will be discarded.

So if so many packet is lost for a time, and UpLinkCounter increase more than MAX_FCNY_GAP. Then end-node will never communicate with server.
Do you think my hypothesis is available? Thanks!

mluis1, 2017-01-13:
For security reasons the UpLinkCounter should always be incremented. Avoids replay attacks.
In LoRaWAN 1.1 version of the specification the retransmissions with same UpLInkCounter have been removed for the same reason.

ckt1010, 2017-01-14:
Hi @mluis1 ,
Thanks for your reply.
But do you think that "never communicate" problem will happen? If so, that will be a big problem too.

mluis1, 2017-01-24:
If you have 16384 frames that aren't received it means that you have something going very wrong in your network.

mluis1, 2017-02-03:
Can we close this issue?

ckt1010, 2017-02-07:
Sorry for later answer.
I agree your explain, but should we just leave it alone?
Maybe use rejoin after too many ack lost is more available

mluis1, 2017-02-09:
The current implementation is compliant with the LoRaWAN specification.
If you want to execute a Join procedure this is application layer responsibility.

closed 2017-02-09 by mluis1

ckt1010, 2017-02-10:
Thanks! I understand now.