Lora-net/LoRaMac-node #180: I receive invalid data when I set fix length payload.

hamidkavianathar, 2017-01-18:
Hi 
I want to send some data with sx1276 and receive it. I am using the semtech library. when I use the variable payload length ( explicit mode ), it works  it works fine but when I set it in fix length (implicit mode), it does not work any more. 
could you please tell me what I should do?
thanks.

mluis1, 2017-01-24:
The class_b branch of this project has an example on how to handle fixed length payloads as the LoRaWAN Class B beacon uses a fixed length payload.

You may also set the ping-pong example to use fixed length payload.

Please note that when using the fixed length payload the settings on the transmitter and the ones on the receiver must match (including preamble length).

mluis1, 2017-02-03:
Can we close this issue?

hamidkavianathar, 2017-02-03:
@mluis1 thank you very much for your answer. finally it worked well. but I have another problem. it is mentioned in datasheet that the bandwidth can be one of these numbers ( in LoRa mode) : 7.8KHz, ... 125KHz, 250KHz, 500 KHz. but the library says the only 125KHz, 250KHz and 500 are valid. I tested the 62.5KHZ and I worked fine but the device didn't work for lower values of bandwidth. could you please tell me, how I can use these values for bandwidth?
thank you.

hamidkavianathar, 2017-02-03:
I apologize for answering late, the github's emails had been gone to my spam folder.

mluis1, 2017-02-03:
The drivers examples have been optimized for LoRaWAN usage. For LoRaWAN only 125, 250 and 500 kHz bandwidths can be used.

If you want to use the other bandwidths you need to modify the drivers accordingly.

Please note that when using bandwidths lower than 62.5 kHz you need to use a TCXO because when the bandwidth gets narrower more the radio is sensitive to RF frequency misalignment.

Please also note that for lower bandwidths you need to enable the bit LowDatarateOptimize.

hamidkavianathar, 2017-02-04:
@mluis1 thanks a lot. your answer helped me a lot. I have another problem SX1276. I am testing the transmitter. with a timer I am sending a constant data with a constant period. in the beginning it works well but after a short time ( about 10 minutes ), it stops sending. could please tell me what I should do?
thanks.

hamidkavianathar, 2017-02-04:
I am working in 500 MHz and after a while I see a signal in 440 and 870 MHz which have high power ( more than 10 dBm). I don't what happens to the SX1276 chip. I am sure that the my MCU is working correctly ( I checked my the SPI lines with oscilloscope).

mluis1, 2017-02-09:
As the issue you are reporting isn't related to the LoRaWAN implementation could you please send your questions to support_rf@semtech.com

closed 2017-02-09 by mluis1

kokroo, 2023-10-14:
> The drivers examples have been optimized for LoRaWAN usage. For LoRaWAN only 125, 250 and 500 kHz bandwidths can be used.
>
> If you want to use the other bandwidths you need to modify the drivers [...]

@mluis1 

Hello, thanks for posting this information. The Zephyr Project uses this module for their LoRa interfaces. I'm working on a Lora project where the bandwidth needs to be set to 62.5 and 31.25 kHz. My board has a TCXO. I created an issue on the Zephyr Project repo and they pointed me here.

While I can note this module is optimized for LoRaWAN, I think it should be possible to set the bandwidth even lower if needed. Is it possible to create that change upstream here so that the Zephyr Project can also utilise it? It's not an API breaking change as far as I can see. Thanks!

mluis1, 2023-10-16:
Recently we provided an answer to a similar request under https://github.com/Lora-net/LoRaMac-node/discussions/1566 discussion.

Not sure how easy is the integration of the [v5.0.0-branch](https://github.com/Lora-net/LoRaMac-node/tree/v5.0.0-branch) changes to the Zephyr project as the new radio drivers have been completely refactored.
At the moment we have no fixed date to get [v5.0.0-branch](https://github.com/Lora-net/LoRaMac-node/tree/v5.0.0-branch)  merged to the main branch and released.

kokroo, 2023-10-17:
@mluis1 Is the 5.0.x branch considered stable? If yes, I can try to initiate a discussion with the Zephyr devs to incorporate it.