Lora-net/LoRaMac-node #182: RTC calibration issue

BBoulet, 2017-01-19:
Hello,

I would like to know what is the purpose of CalibrateSystemWakeupTime() function called in BoardInitMcu. Sometimes (~10% chances) my system remains asleep and never wakes up, this is quite problematic.
Does anybody experienced this ?

GregCris, 2017-01-25:
The function CalibrateSystemWakeupTime() is used to measure the time taken by the MCU to wake-up form stop mode and re-initialise the various peripherals (SPI, I2C, ...). The time varies from one HW to another due to the MCU Xtal (if used), the core clock speed, ....

mluis1, 2017-02-03:
Can we close this issue?

closed 2017-02-09 by mluis1

dmeehan1968, 2021-01-20:
I appreciate that this is an old issue, but the RtcSetMcuWakeUpTime implementation for STM32L0xx doesn't appear to work.  When this function is called 1 second after MCU initialisation is complete, the function fails to do the computation as `HAL_NVIC_GetPendingIRQ( RTC_IRQn ) == 1` is not true.  As a result, the global `McuWakeUpTimeCal` is left at 0.

The additional startup delay (1 sec) is redundant and unhelpful.

Also, the sample apps always set USB power connected, which effectively disabled low power modes, so this code gets skipped.  My use case is battery power, and so the startup delay is unhelpful.

altishchenko, 2021-01-20:
@dmeehan1968 Hello Dave, leaving this value at 0 doesn't break anything in my devices running on L051 with LiPo's and using deepsleep extensively. I actually dropped this logic from my firmware some good time ago to save on space and remove unneeded complexity.

mluis1, 2021-01-20:
While discussing with ST sometime ago they also said that they removed this logic from their port.

We will provide an update removing this logic as soon as possible.

I have created a new issue in order to not forget to this update.