Lora-net/LoRaMac-node #183: Modularized Regional Support

loverdeg-ep, 2017-01-20:
> 1 1 Introduction
> 2
> 3 This document describes the LoRaWAN™ regional parameters for different regulatory
> 4 regions worldwide. This document is a companion document to the protocol specification
> 5 document, LoRaWAN Specification [LORAWAN], starting with version 1.0.2 of the
> 6 specification. Separating the regional parameters from the protocol specification allows the
> 7 addition of new regions to the former without impacting the latter document.

Separating the regional parameters implementation from LoRaMac.c & LoRaMac.h should allow addition of new regions to the former without impacting the latter as well.

Sorting through the if defines for each region really slow me down when reviewing the source.

mluis1, 2017-01-23:
We are currently finalizing the 1.0.2 implementation. And the stack architecture has been changed as your proposition. Each region will have it's own files.

closed 2017-01-23 by mluis1

loverdeg-ep, 2017-01-24:
How much will the "API's" change with the next release?
I've been considering merging some of the dev branch into a fork of the Mbed fork but won't waste time if there will be a lot of changes.

mluis1, 2017-01-25:
We will synchronize the mbed libraries with GitHub once we have finished the release.
https://developer.mbed.org/teams/Semtech/code/LoRaWAN-lib/ will be synchronized with master branch.
https://developer.mbed.org/teams/Semtech/code/LoRaWAN-lib-dev/ will be synchronized with 1.0.2 branch. (Regional parameters support.)

The changes will not be big. Some APIs will have an additional parameter in order to specify the region.

The biggest change that will happen on mbed is the timers time base. Currently they are based in us and with future release they will be ms based. By doing this we will simplify the synchronization between GitHub and mbed.