Lora-net/LoRaMac-node #184: Block the MCU from going into Low Power mode

spiras, 2017-01-23:
* \brief This function blocks the MCU from going into Low Power mode
void BlockLowPowerDuringTask ( bool status );

However, in the default implementation of RTC_Alarm_IRQHandler
there is a
`BlockLowPowerDuringTask(false);`
which actually removes the desired block.

Is this intentional?
If yes, how should it be possible to keep the MCU awake during a task?

GregCris, 2017-01-25:
The enabling or disabling of the Low Power mode is controlled by the variable status:
When status is true, the MCU will not go into low power mode
When status is false, the MCU will go into low power mode (Arm STOP mode)

spiras, 2017-01-25:
Assumed we all know how functions are controlled by variables, please notice I am reporting an issue: it is not possible to keep Low Power Mode disabled, because it is enabled again by RTC_Alarm_IRQHandler every time a timer is used.

GregCris, 2017-01-25:
"it is not possible to keep Low Power Mode disabled, because it is enabled again by RTC_Alarm_IRQHandler every time a timer is used". 

>>This is not an issue, this is made on purpose! The whole concept is to be in Low power mode as much as possible and to only disable the low power mode for certain specific functions. If the low power mode was not needed, we could have used a much simpler implementation. If your system is powered from a continuous power source, you are free to change the code for your system.

spiras, 2017-01-25:
Yes I am on battery. Yes I need to keep low power mode disabled only for a certain specific function. And I need timers for that specific function. But timers switch low power mode on again, so the specific function cannot be accomplished, because the MCU goes in Low Power mode instead

spiras, 2017-01-25:
I suggest to remove the following line:
`BlockLowPowerDuringTask(false);`
from the default implementation of `void RTC_Alarm_IRQHandler( void )` in rtc-board.c

mluis1, 2017-02-09:
Can we close this issue?

spiras, 2017-02-13:
The only way I found to make the system work as intended, is to revome a line of code from the default implementation, as explained above. So I can call the BlockLowPowerDuringTask(false) only when the task is finished, and let it go back in low power mode. Otherwise, it keeps going in low power mode before the task is finished. Can you please confirm or correct me, before we close the issue?

GregCris, 2017-02-24:
Sounds good to me. As long as you can enter and leave low power mode when needed, there is no reasons this could be an issue. We provide an example code which is optimized for our implementation, then it is down to the user to select when to enter or leave STOP mode depending on the situation.

spiras, 2017-02-28:
As you say, it should be down to the user to decide when the critical task is finished, but currently it is NOT. 
After calling BlockLowPowerDuringTask(true), the user expects that lowpower remains disabled until he calls BlockLowPowerDuringTask(false) at the end of the task. But this is not how it is working.
For example, if during the task any other timer elapses (i.e. led blink), the system will silently re-enable lowpower, and the critical task may fail with apparently no reason, and this may be hard to debug.
That said, please close this topic at your own discretion.

closed 2017-03-06 by mluis1