Lora-net/LoRaMac-node #185: Maximum number of trials in JOIN - Part 2

sda3422, 2017-01-24:
I would like to re-discuss about the issue #156 closed.
@djaeckle : Thank you for taking into account the need in this issue. However, I have few remarks.
The idea was also to try to take into account on field installation by a human waiting for an installation status.
First, in the function LoRaMacMlmeRequest you force Req.Join.NbTrials = 48 minimum so, if I set this value to 3 it is overwritten (no really useful then).
Then, the AlternateDatarate is designed to work with this kind of high value. With a 10 seconds delay between the tries, the installation guy on field will wait a very long time if the unit is only accessible through DR0 (worst case of course).
It could be very valuable to be more flexible on the join procedure to take into account different way of working (low power, installation by a human waiting a feedback, ...).
What do you think ? Thanks.

mluis1, 2017-01-24:
The Join procedure must follow the specified back-off constraints. (LoRaWAN specification chapter 8)
The `AlternateDatarate` function ensures that the implementation fulfills the chapter 8 constraints.

Thus, in order to give a chance for the lower datatarate (SF12) to be used at least one time, the minimum amount of attempts to be performed is 48.

The 48 attempts should only happen in case the environmental conditions of the node are really bad. The way the implementation is done the node shall connect to the network using the highest possible datarate.

By setting the number of trials to 3 as in your example the Join procedure attempts will be done 3 times at SF7 which implies that you need to have the best possible environment for the node. In case the node is placed for instance in an environment that requires SF8 it will never get connected to the network.

sda3422, 2017-01-24:
@mluis1 Ok I understand your point of view. We have to review the water metering product installation and consider only "automatic" installation and not installation with "immediate feedback" as we did in the past.
Regarding this part of the code in the function `LoRaMacMlmeRequest` :
_// Enables at least the usage of all datarates.
if( mlmeRequest->Req.Join.NbTrials < 48 )
{
                mlmeRequest->Req.Join.NbTrials = 48;
}_
Setting `mlmeRequest.Req.Join.NbTrials = 3` is always overwritten with value 48.
Is it really necessary ? If I want to set less trials (even if all datarates are not used) I can't.

mluis1, 2017-01-24:
The problem that I have in removing this condition is that in the future we will have users complaining that their node isn't able to connect to the network because they have put a number of trials lower than 48.

As you are aware of the potential issues please feel free to comment out this condition on your implementation.

sda3422, 2017-01-24:
Ok. One linked question (I try to find information on the web but nothing clear) and I close the issue : if I build my lorawan private network, do I have to follow the values set by the lorawan spec ? Like ADR_ACK_LIMIT, etc ? Of course assuming that I am fully compliant with the ETSI rules ! Thanks.

mluis1, 2017-01-24:
If you intend to connect your devices to a public network in the future then yes, you need to comply with the LoRaWAN specification values.

sda3422, 2017-03-10:
Hi,
@mluis1 and @djaeckle , just merging last stack in my project, I am surprised that finally we cannot set an "infinite" number of join requests by setting `mlmeReq.Req.Join.NbTrials = 0`.
Isn't it interesting to be able to do it without the support of the application ?

mluis1, 2017-03-17:
As of today it is easier to let the application handle the infinite number of trials.

In case the MAC layer reports that the Join procedure failed the application just needs to repeat the process up until the Join is successful.

closed 2017-03-19 by sda3422