Lora-net/LoRaMac-node #186: RX1 reception overlaps RX2 timer event - Bad RxSlot

sda3422, 2017-01-26:
Hi, I found an issue on v4.3.0. Sorry I cannot test on develop branch but I try to find this issue in the list and nothing appears.

**Configuration :**
CLASS A
Uplink -> TX DATARATE = DR0
Downlink -> ON RX1

**Issue :**
The issue observed on reception of a downlink frame is : `mcpsIndication->RxSlot = 1` in MCPS-Indication event function instead of `mcpsIndication->RxSlot = 0`.

**Potential explanation :**
The length of the downlink frame is higher than the delay between RX1 and RX2 because of the use of DR0.
In that condition, `OnRxWindow2TimerEvent` is called before `OnRadioRxDone`, the result is `RxSlot = 1`. The good thing is that the execution of `RxWindowSetup` function called from `OnRxWindow2TimerEvent` is "empty" because of the test : `if( Radio.GetStatus( ) == RF_IDLE )`.

mluis1, 2017-01-27:
Thanks for the report. 
Your observations are right and the same issue can be found on develop branch.
We need to think about a fix for this.

While we don't find a better fix I propose the one described below. 
Please try to modify the OnRxWindow2TimerEvent function as follows:
```C
static void OnRxWindow2TimerEvent( void )
{
    uint16_t symbTimeout = 5; // DR_2, DR_1, DR_0
    uint32_t bandwidth = 0; // LoRa 125 kHz

    TimerStop( &RxWindowTimer2 );

    if( Radio.GetStatus( ) == RF_IDLE )
    {
        RxSlot = 1;
    }

...

}
```

sda3422, 2017-01-27:
Thanks for the feedback.
I am ok with this first fix.

closed 2017-02-09 by mluis1