Lora-net/LoRaMac-node #187: mlmeConfirm never called. Device stuck in sleep more after OTAA

LaurentHB, 2017-02-10:
Hello,

I am using your API to connect a device to a multitech gateway. I have been able to get ABP working just fine. I am now trying to make the device join the gateway via OTAA and am running into some problems.

I can see the device joins correctly and the gateway is sending back the confirmation but my device never goes out of SLEEP state. My limited understanding of the API tells me it never went through mlmeConfirm to change the state to SEND.  Do you have any suggestion to find what isn't working with the code? I don't quite know where to start or what I'm looking for.

Thanks!

djaeckle, 2017-02-15:
Hi LaurentHB,

are you using the sample application or a modified one? Which hardware platform are you using? Which branch/commit are you using?

LaurentHB, 2017-02-15:
I'm using a sx1276 and a MLK17Z256VFM4, a colleague ported the drivers used in a demo to our system. I basically only changed the payload that is being sent. 

I am currently getting the interruptions from the DIO_0 to signal  Tx is done when I'm in that mode but I never get the same interruption from DIO_0 to signal Rx is done. Do you have any idea of how I could go about debugging this? It could be a problem with our system but I'm not so sure about how to go about seeing what is causing the problem.

Sorry if it's not really precise information, I'm an intern and have a very limited understanding of how all of this works. 

Thanks for the help!

djaeckle, 2017-02-16:
Hi LaurentHB,

thanks for the feedback.

> I can see the device joins correctly and the gateway is sending back the confirmation but my device never goes out of SLEEP state.

> I am currently getting the interruptions from the DIO_0 to signal Tx is done when I'm in that mode but I never get the same interruption from DIO_0 to signal Rx is done.

It seems like the timer is not running correctly. The timer is responsible to wake-up the system from sleep mode. 
Regarding your use case, it could be that:

- The node sends an uplink
- The node tries to setup the timers for the RX windows
- The node goes to sleep
- The node stays in sleep mode because it never gets a timer interrupt to wake up

I would double check the platform specific timer implementation.

LaurentHB, 2017-02-16:
Hello djaeckle, 

Thanks you for the answer.

I looked further into the problem and it seems my gateway is not responding to the join request. The problem, at least for now, doesn't seem to be coming from LoRaMac-node. I'll keep in mind what you said in case I run into this problem again. I have another question but since it's not directly related to this, you can close this issue and I will open a new one later.

LaurentHB, 2017-02-20:
Hello,

I'm actually reopening this issue because I fixed whatever was happening on the gateway's side. I'm still trying to receive a packet without success.

If I understand the sx1276 specs correctly, before receiving the actual packet, the chip will go through channel activity detection. When it's done, it will raise the DIO3 ( given the right DIO mapping, and that should be the case for LoRaMAC-node) to indicate its done checking for activity and if it caught a lora preamble or not. 

When setting breakpoints in my code, my DIO3 is never raised. That makes sense that  I would never have a RxComplete because my device was never really waiting for any packet anyway. 

What do you think can be done to know why the DIO3 is never raised? Or to know the CAD is at least starting?

Thanks!

closed 2017-02-23 by LaurentHB