Lora-net/LoRaMac-node #190: ChannelsNbRep for join retries

dudmuck, 2017-02-13:
ResetMacParameters() called from OnTxDelayedTimerEvent() causes max join retries to be limited to two tries because LoRaMacParams.ChannelsNbRep is reset to default value of 1.

djaeckle, 2017-02-14:
Hi  dudmuck,

thanks for the report. I cannot reproduce the issue. Could you provide some more details?

dudmuck, 2017-02-24:
Its not band-specific.  You can see this when join-request is not accepted, so it might be quicker to try on region which doesnt have duty-cycle limiting. 
Check that ChannelsNbRep reaches desired number of join attempts when server doesnt give join-accept.  This could be tested by using wrong DevEUI or aes-key or shutting off gateway.  The desired number of attempts is specified in NbTrials in the MLME_JOIN request from application layer.

closed 2017-02-28 by mluis1