Lora-net/LoRaMac-node #191: How to set the 470MHz band

zhoubx123, 2017-02-15:
I set the 470MHz from 433MHz，and just change #define LC1 433175000 to 471300000;#define LC2 433375000 to 471500000;#define LC1 433575000 to 471700000;
it works，However, after the ADR is  on, the SF7 and SF8 are unable to receive the gateway Ack.
Where do I need to set?

djaeckle, 2017-02-15:
Hi zhoubx123,

you can configure the stack to use the CN470 band if you define the symbol `USE_BAND_470`. You need to define it in the preprocessor options of your environment.

zhoubx123, 2017-02-15:
I have not define USE_BAND_470，and just USE_BAND_433,
at SF12 to SF9, module can receive gateway Ack, but SF8 and SF7 can not;
and gateway can receive date from module.
so I do not know some where must be set?

djaeckle, 2017-02-15:
Do you want to use the EU433 or the CN470 band?

zhoubx123, 2017-02-16:
no, i just use CN470 band.

zhoubx123, 2017-02-16:
I would like to use the 433MHz definition, but running in 470MHz, modify the following（LoRaMac-definitions.h）:
/*!
 * LoRaMac default channels
 */
// Channel = { Frequency [Hz], { ( ( DrMax << 4 ) | DrMin ) }, Band }
//#define LC1                { 433175000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
//#define LC2                { 433375000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
//#define LC3                { 433575000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC1                { 471300000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC2                { 471500000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC3                { 471700000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC4                { 471900000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
Do I need to set any place?

djaeckle, 2017-02-16:
Hi zhoubx123,

have you also updated the RX2 window definition ( `RX_WND_2_CHANNEL` )?

Please note that your implementation will not be compliant to the LoRaWAN Specification as this regional definition is not specified.

zhoubx123, 2017-02-17:
yes, I have:
/*!
 * Second reception window channel definition.
 */
// Channel = { Frequency [Hz], Datarate }
#define RX_WND_2_CHANNEL                                  { 478500000, DR_0 }
I just want to test, so  have not define CN470 band.
it can work, but SF7 and SF8, module can not receive gateway ACK.

djaeckle, 2017-02-24:
Hi zhoubx123,

those adaptations are everything you need. Unfortunately, we cannot provide more support on your issue, as your regional definition is not compliant nor specified in the LoRaWAN Specification.

closed 2017-02-28 by mluis1