Lora-net/LoRaMac-node #192: Support of higher grade STM32L1 MCU

abouillot, 2017-02-21:
I'm currently trying to use the LoRaMac node on STM32l152 Nucleo board.
I've migrated the project to Keil µVision V5, as this MCU is not supported on V4 ( stop at L152RD). This has required few changes, as the `RCC_IT_LSECSS` correction as well as `AHBPrescTable` and `APBPrescTable` declaration in RCC HAL file.

Unfortunately, this is not sufficient. It seems the timer interrupts are not raised. I get the first message sent, but not a second one.

Has anybody have an idea of what should be donne additionally?

Thanks

nestorayuso, 2017-02-21:
ST has already ported the LoRaMac-node code to more STM32 device families

http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-expansion-software/i-cube-lrwan.html

abouillot, 2017-02-21:
Yes, but this is using a different HAL, and I'm willing to use the core Semtech version to be on the cutting edge of the version, at least for the time being.

mluis1, 2017-02-21:
We are currently finalizing the addition of the [NAMote72](https://developer.mbed.org/platforms/NAMote-72/) platform which is based on a STM32L152RC MCU.

reixd, 2017-02-21:
I am also interested on running LoRaMac on the STM32L152RE Nucleo boards.

The Semtech iCube LoraWAN port does work with the L152RE but there is apparently a problem with the HAL driver and the (RTC) timer. The serial UART interface only works by sending bytes to the PC, but it is unable to receive any bytes from the PC. All HAL UART methods did not seem to work here, because the timer variable on the semtech driver code is not being incremented, thus any timeouts cannot be reached, if the timer always equals 0....
I wrote a description of this problem here: https://community.st.com/thread/36638-cannot-rx-serialuart-data-using-i-cube-lrwan . Maybe somebody may have an idea on how to fix the problem.

abouillot, 2017-02-21:
@mluis1 L152RC is where things get wrong, as for RD and RE. When will the initial code will be available?

mluis1, 2017-02-21:
I Have just pushed the NAMote72 platform to the develop branch.

I close the issue.

closed 2017-02-21 by mluis1

abouillot, 2017-02-21:
@mluis1 Thanks it solved my issue! I'll digg further to have a clean port, but changing the rtc code from SensorNode to NAMote72 fixed the timer issue.

reixd, 2017-02-22:
@abouillot would you publish or upload your code/solution?

abouillot, 2017-02-22:
@reixd you can find it @ https://github.com/abouillot/Nucleo_LoRaMac

reixd, 2017-02-24:
@abouillot Thx :+1: