Lora-net/LoRaMac-node #196: Question about RX windows

mpt8153, 2017-02-22:
I've readed issue [https://github.com/Lora-net/LoRaMac-node/issues/181](url).
In AN1200.23 from semtech at page 7 is written "_the receiver requires 5 symbols to detect preamble and  synchronize_".
I've a doubt about Rx window timeout for lower datarates (DR0, DR1, DR2). In this implementation is setted to 5 symbols:
`static void OnRxWindow1TimerEvent( void )
{
    uint16_t symbTimeout = 5; // DR_2, DR_1, DR_0
    int8_t datarate = 0;
    uint32_t bandwidth = 0; // LoRa 125 kHz

    TimerStop( &RxWindowTimer1 );
    RxSlot = 0;
.......`

I think it should be setted to 6 in order to take account of deviations of 0.488ms of RTC tick.
What do you think?

dsmith-tek, 2017-02-22:
If you take a look at AN1200.23 ( [PDF](http://www.semtech.com/images/datasheet/SX1272_settings_for_LoRaWAN_v2.0.pdf) )there is a really good discussion of RX Window timing.

The suggested schema uses both the symbol timeout AND a RX Offset for each DR to maximize the preamble overlap. The current stack doesn't easily allow for the implementation of a DR based RxOffset however.

Based of the analysis in AN1200.23 I agree with you. I ran the numbers for 0.5msec error and while it suggests a symbTimeout of 5 is adequate, all of them have a positive RX Delay. This implies the RX windows should be opened late (1 to 12 msec depending on DR) in order to prevent the window from opening before the gateway starts transmitting.

Since my current Dev platform doesn't need to be energy efficient I grossly over extend my windows (and open them early) to eliminate any timing error. Even with these very liberal Rx windows times however, I still see downlink packet loss (~6 to 10%).

desolatormkx, 2017-02-22:
If you take a look at AN1200.23 ( [PDF](http://www.semtech.com/images/datasheet/SX1272_settings_for_LoRaWAN_v2.0.pdf) )there is a really good discussion of RX Window timing.

The suggested schema uses both the symbol timeout AND a RX Offset for each DR to maximize the preamble overlap. The current stack doesn't easily allow for the implementation of a DR based RxOffset however.

Based of the analysis in AN1200.23 I agree with you. I ran the numbers for 0.5msec error and while it suggests a symbTimeout of 5 is adequate, all of them have a positive RX Delay. This implies the RX windows should be opened late (1 to 12 msec depending on DR) in order to prevent the window from opening before the gateway starts transmitting.

Since my current Dev platform doesn't need to be energy efficient I grossly over extend my windows (and open them early) to eliminate any timing error. Even with these very liberal Rx windows times however, I still see downlink packet loss (~6 to 10%).

closed 2017-03-17 by mluis1