Lora-net/LoRaMac-node #197: RxContinuous FIFO Data Corruption

Brulath, 2017-02-22:
Utilising the radio portion of the code base in RxContinuous mode will result in recycled data being produced after 256 bytes have been received.

I sent packets containing 5x 32-bit values. After exhausting the buffer I would receive a packet from the radio which included the last four bytes of one packet and the remaining four from another. So all message received below the buffer size would be:
```<1> <2> <3> <4> <5>```
Then after the buffer was exhausted by 4 bytes the messages would arrive as:
```<5 (prev)> <1> <2> <3> <4>```
Later it would become exacerbated:
```<4 (prev)> <5 (prev)> <1> <2> <3>```

Putting the radio to sleep and then initiating a new Rx after receiving each packet entirely eliminates the issue. So one of the FifoRxCurrentAddr, FifoRxBaseAddr, or FifoAddrPtr registers are losing sync with the remaining two. Actions that appear to reset them correct the issue.

GregCris, 2017-02-24:
Hello Brulath,
The addresses FifoRxBaseAddr and FifoAddrPtr are reset to 0 in the function "SetRx()" (please, see lines 884 and 885 in sx1272.c and lines 1037 and 1038 in sx1276.c). The pointer FifoRxCurrentAddr is read only and it is directly controlled by the radio. In your case, I understand that you are in Rx Continuous mode, are you leaving the Rx mode between packets?

Brulath, 2017-02-24:
The sx1272 remained in continuous receive mode, and did not perform any action outside of receiving packets, when the issue occurred. The intention was to have the device only receive and transfer packets to another device, so it was just going to sit waiting for a packet received interrupt.

Unrelated, I had thought during my experiments to discover the issue that I was able to write to all three registers, despite the documentation indicating otherwise. I don't recall whether doing so proved promising, though, as we decided to avoid continuous receive instead whilst it appeared to be unreliable :)

GregCris, 2017-02-27:
I can assure you the Rx Continuous mode works reliably :-) 
The issue here is more related to the drivers than anything else The drivers are optimized for the LoRaWAN and not necessary for your own implementation. You could easily fix the issue by Resetting the FIFO pointers just after a reception, at the end of the function SX1272OnDio0Irq(), just after you have read the FIFO.

Brulath, 2017-02-27:
That is true. You may find it to be an issue with this code base when you begin implementing Class B. Thanks for confirming the resolution :)

closed 2017-02-27 by Brulath

FabianInostroza, 2018-08-04:
How to correctly reset the internal rx pointer?

GregCris, 2018-08-07:
There are two registers called RegFifoRxBaseAddr and RegFifoAddrPtr. You need to set the value for RegFifoAddrPtr at the same level as RegFifoAddrPtr and start reading the payload from RegFifoRxBaseAddr.