Lora-net/LoRaMac-node #198: Corrupted frames in high SF modes

quentin-ol, 2017-02-23:
Hello everyone, 

We're experiencing very high PERs on high SF modes on a sx1276 chipset (we use the Mbed SX1276MB1MAS) using the class A example.

This issues affects RX1 and RX2 windows on all channels and is more frequent  the higher the SF.

In an attempt to narrow down the issue, we statically configure the RX1 window like this:

```
Radio.SetRxConfig(
            MODEM_LORA,
            RX_BANDWIDTH, // 0
            rx1DataRate, // 12
            RX_CODE_RATE, // 1
            BANDWIDTH_AFC, // 0
            RX_PREAMBLE_LENGTH, // 8
            RX_SYMBOLS_TIMEOUT, // 20 (tried many values between 5 and 20)
            PAYLOAD_FIXED_LENGTH, // 0
            0, 
            RX_CRC, // 0
            RX_FREQUENCY_HOPPING, // 0
            RX_FREQUENCY_HOPPING_PERIOD, // 0
            RX_IQ_SIGNAL_INVERTED, // 1
            RX_CONTINUOUS // 0
    );
Radio.SetMaxPayloadLength(MODEM_LORA, MAX_PAYLOAD_SIZE); // 255
Radio.Rx(RX_DURATION_MAX); // 3000
```
This configuration is setup `RADIO_WAKEUP_TIME` ms before RX1_DELAY to account for the oscillator startup time and the transition from sleep to RX state.

The issue is seen in the RXDone callback. The first bytes of the payload are often correct (usually the corruption shows up around the FrameCnt bytes), _e.g._:
[image]

 - the frameCnt is supposed to be {0x00,0x00}, not {0x00, 0x80}.
 - the MICs do not match : 0xd6054d01 & 0x4cdea041 (locally computed)

By the way, the RSSI reported by the radio is in line with what's expected: -88dBm but the SNR is very low: -16 (device configured to use SF12).

Is there anything obvious that we might be overseeing ? Anyways, any help or feedback is welcome.

On a side note, we have an almost (this issue !) working BSP for the Nucleo L152RE/ SX1276MB1MAS board which we will gladly share when this issue is fixed.

GregCris, 2017-02-24:
Hello Quentin,

Quick questions, do you see the same behavior with other SF or is it only with SF12? Are you using the standard drivers provided by Semtech or have you changed something?

quentin-ol, 2017-02-24:
Hey Greg,

The issue happens all the time with SF12. We've witnessed a few corruptions with SF11 but only a few times. 

Regarding drivers, we're using the sx1276.c/h radio.h files of this repository.

quentin-ol, 2017-02-24:
Update&closure: the issue is in our example code.

The example we used to investigate the issue was an oversimplification that caused the corruption.
In short, we scheduled the Rx1 and Rx2 windows one second apart using 2 timers. The OnRxWindow2 event setup the RX2 window without checking the radio state. With SF12, the radio was still receiving (RF_RX_RUNNING) when configuring the radio for the RX2 window.

Although this does not address the high PERs that we witness, this issue is no longer meaningful. 

Sorry for the inconvenience.

Quentin

closed 2017-02-24 by quentin-ol