Lora-net/LoRaMac-node #199: SX1276 mode register going straight to standby instead of Rx single

LaurentHB, 2017-02-23:
Hello,

I made a new issue and closed the old one because of the more precise question but it does refer to basically the same problem so just let me know if you'd rather have me add it to the end of the old one instead and I will gladly do it.

I'm currently having trouble with the Rx portion of my device running LoRaMAC-node. I am using an SX1276 and I am finding a pretty strange behavior with specifically the RegOpMode register.

I read the register right after the SetOpMode function is called to make sure  the Mode bits are correctly set to 011 for TX and 110 for RXSINGLE.  I am finding that right after the bits are written for TX, the packet is sent to the gateway correctly and when I read the register right after, it already shows the radio is in STDBY.  This seems to be normal behavior since the radio automatically goes to STDBY when the TX done interrupt is called. 

The problematic part is that when I do the same procedure after putting the device in RXSINGLE mode, I also see the device in STDBY mode when I read the register right after setting it. This doesn't seem to be normal behavior since it should only automatically go into STDBY mode if the RxTimeout is called or if RxDone is called which are, respectively,  3s and minimally 1s after the register is set.

Do you have any idea why this might be happening? I'm definitely running out of idea to pinpoint what the problem might be.

Thanks a lot!

GregCris, 2017-02-24:
Hello Laurent,
I don't think this is an issue and this behavior seems normal to me.
Let me explain: when you set the radio into Tx or Rx mode from the standby mode, the radio has to perform several action before actually being in Rx or Tx mode (switching the XTAL, locking the PLL, starting the packet controller or starting the PA,... ) This takes a little bit of time (around 250uS from the top of my head but you can check the numbers in the datasheet). So if you are sending an SPI read command straight after setting the device in Rx, the state machine will indicate it is still in Standby while the radio is actually processing the command. You can try adding a small delay before reading back the OpMode register and all should then be as expected.

LaurentHB, 2017-02-24:
Hello Greg,

Thanks for the answer. I did think that might be the case so I tried adding up to a few mS of delay before reading the  register again and come up with the same result. The register will still indicate 0x89.

I'm kind of running out of ideas to see what is happening with the sx1276 when I'm trying to receive with it. Maybe you've been through similar problems, so do you know any way to debug it that would point towards what might be happening wrong? It's kind of guess work at this point and it not very effective unfortunately. 

Thanks again.

kwmartin, 2017-03-21:
1) Whenever a packet is correctly received, OnRadioRxDone() in LoRaMac.c is called from SX1276OnDio0Irq() in sx1276.c. The first thing OnRadioRxDone() does is to put the sx1276 to sleep. This occurs for both Continuous and Single Mode so you don't need to turn Rx off in Continuous Mode, it occurs normally.
2) Single Mode is stopped by an sx1276 time out based on what is programmed in bits 1:0 of 0x1E and also register 0x1F. These 10 bits give a maximum time out 1023 symbols (each symbol is (2^SF)/BW). If bit 7 is not set in 0x12, then SX1276OnDio1Irq() is called which calls OnRadioRxTimeout() in LoRaMac.c; if bit 7 is set, the IRQ is masked out, and SX1276OnDio1Irq() is not called, but SingleMode still stops after the programmed number of symbols and the sx1276 goes to Standby (0x1 = 0x81 or 0x89 - bit 3 - LowFrequencyModeOn is irrelevant as far as the data sheet is concerned). Your problem may be the sx1276 is missing the preamble during the time Rx is active. I am also seeing it missing receives after sx1276 reset even when I know it hits the preamble (between 20-60% of the time), but not if a TX occurs first (I think - still working on it).
3) Coming from reset of the sx1276 and going directly to Rx, I am seeing sx1276 timeout occurring way too early - after either 39ms or 62 ms. I think this does not occur if I do a transmit first, but am not sure yet. I have not yet been able to track this problem down. For this reason, I am also leaning towards using the software timeout. This resolves the Timout issue, but not the spurious receive issue after reset.
Summary: using RxSingle after sx1276 reset seems to have problems that are not really addressed in the LoRaMac code. If anyone has additional info here, it would be much appreciated.
4) I think the MacStateCheckTimer is LoRaMac's way of replacing os_runloop() from IBM's LMIC oslmic.c.  MacStateCheckTimer wakes up the CPU once a second when it is active and sets event flags. My initial impression is this could use some improvement, but I'm reserving judgement for now. My guess is both of these should be replaced by an RTOS, but the issue is how to ensure everything is being done correctly without having to continuously wake from Stop Mode which limits battery life. Maybe what is really needed is an RTOS with special features for waking from Stop based on Alarm Timers? Is there any activity to be compatible with LoRaWAN without using either LMIC or LoRaMac? or to correct the limitations of one of these?

LaurentHB, 2017-03-24:
Easiest workaround is just to manage the windows via MCU timers. It looks like a problem that might be very specific to my implementation so I will close this issue.

closed 2017-03-24 by LaurentHB