Lora-net/LoRaMac-node #200: How to add a RU868 band?

Cr0s, 2017-02-27:
Hello.

I'd like to know how to modify definitions in LoRaMAC code in order to add a new frequency band named RU868 (Russian 868 MHz band, LoRa Alliance approval is pending).

The band's characteristics are following:
[image]

Cheers!

mluis1, 2017-02-28:
So far I wasn't aware of a Russian band definition by the alliance. But, when they add it to the Regional parameters document we will provide the support for it.

Concerning your changes we think that you have covered almost all the aspects. Maybe the one that is left is searching everywhere the USE_BAND_868 definition usage and add the USE_BAND_868_RU condition as well.

closed 2017-02-28 by mluis1

nestorayuso, 2017-02-28:
Here are some drafts of regional parameters for Russia
https://github.com/brocaar/loraserver/issues/113

mluis1, 2017-02-28:
@nestorayuso Thanks for the info.